riscv-software-src/riscv-isa-sim #1037: Dumping memory signature

abdullahyildiz, 2022-07-04:
Does the [release v1.1.0](https://github.com/riscv-software-src/riscv-isa-sim/releases/tag/v1.1.0) support dumping memory signature?

There is no such option in this version of Spike binary:

https://github.com/riscv-software-src/riscv-isa-sim/blob/530af85d83781a3dae31a4ace84a573ec255fefa/spike_main/spike.cc#L18-L77

However, the host interface seems to support it:
https://github.com/riscv-software-src/riscv-isa-sim/blob/1767a27ad4cf8f9eb1a30d9d9a2495477071339f/fesvr/htif.cc#L150-L155

Phantom1003, 2022-09-23:
Yes, memory signature still works, use `+signature=` to specify the signature file.
htif will also parse command line arguments, it looks like `-h` conflicts with spike, you can use `+h` instead.
https://github.com/riscv-software-src/riscv-isa-sim/blob/2a7c33b8b50f91379a12f983dd3c1c7ee9f89a90/spike_main/spike.cc#L408-L409
https://github.com/riscv-software-src/riscv-isa-sim/blob/1767a27ad4cf8f9eb1a30d9d9a2495477071339f/fesvr/htif.cc#L341-L344

abdullahyildiz, 2022-09-25:
@Phantom1003,

Could you share the command syntax supporting `+signature=`?

Phantom1003, 2022-09-26:
Try `spike +signature=<signature file> <testcase file>`.

abdullahyildiz, 2022-09-26:
There are other options that I need to add to the command line and the following one didn't work:

`spike -l --log-commits -m0x40000000:0x100000 --extlib=../spike_mmio_device_plugin/plugin.so --device=test_mmio_plugin,0x20000000,argument +signature=test.sig test.riscv`

hakan-demirli, 2023-02-20:
I have encountered a similar issue. Spike either didn't do anything or complained:  ```unknown option "+signature"```. The problem was the position of the "+signature" and "--signature" arguments. Putting them right before the *.riscv file solved the issue.

This one works:
```
spike -l -m0x40000000:0x80000 --isa=rv32imc +signature=test_0178.sig --signature=test_0178.sig bin/test_0178.riscv 2> log/test_0178.log
```
This one works too:
```
spike -l -m0x40000000:0x80000 --isa=rv32imc +signature=test_0178.sig bin/test_0178.riscv 2> log/test_0178.log
```

This one does not:
```
spike -l -m0x40000000:0x80000 --isa=rv32imc --signature=test_0178.sig bin/test_0178.riscv 2> log/test_0178.log
```