riscv-software-src/riscv-isa-sim #1046: Spike throws exception on PMP CSR access

Abdulwadoodd, 2022-07-15:
For the following PMP CSRs access, SPIKE throws an exception:
- pmpcfgx where x is [4,15]
- pmpaddrx where x is [16, 63]

I have written a test where I configure the `pmp16cfg` entry in TOR mode and hence used `pmpcfg4` which in turn uses `pmpaddr16` to `pmpaddr19`. SPIKE throws exceptions on these CSR accesses, unlike SAIL. SAIL is giving the expected results and working fine after [these](https://github.com/riscv/sail-riscv/pull/168/files) changes has made in sail-riscv. 
The test file is attached below and it is run under riscof framework.

[pmp-addr_match-tor_mode.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/9120830/pmp-addr_match-tor_mode.txt)

aswaterman, 2022-07-15:
Not a bug.  The ISA spec says that "implementations may support zero, 16, or 64 PMP CSRs."

The SAIL model was correct both before and after your change.  Spike is correct today.

Abdulwadoodd, 2022-07-15:
True, but do we have support for 64 PMP CSRs in Spike? 
Apparently, `encoding.h` file in this repo contains all 64 PMP CSRs but Spike supports only 16 PMP CSRs.

aswaterman, 2022-07-15:
As a matter of courtesy, please don’t submit feature requests as bug reports.

encoding.h is an auto-generated file that doesn’t relate to what spike supports.