riscv-software-src/riscv-isa-sim #1058: WFI in User mode

aprnath, 2022-08-02:
Seems like Spike throws illegal exception when encountering WFI in user mode, regardless of mstatus.TW.

The priv spec has several different statements, which are seemingly contradictory regarding user mode functionality of WFI. While it is clear that when mstatus.TW=1, WFI is illegal in SU modes, these other statements leave the mstatus.TW=0 behavior somewhat ambiguous for User mode. Excepts below:

3.3.3 Wait for Interrupt
.. _WFI is available in all privileged modes, and optionally available to U-mode_. This instruction may raise an illegal instruction exception when TW=1 in mstatus, as described in Section 3.1.6.5.

3.1.6.5 Virtualization Support in mstatus Register
.. _When TW=0, the WFI instruction may execute in lower privilege modes when not prevented for some other reason._ When TW=1, then if WFI is executed in any less-privileged mode, and it does not complete within an implementation-specific, bounded time limit, the WFI instruction causes an illegal instruction exception. 

.. _When S-mode is implemented, then executing WFI in U-mode causes an illegal instruction exception, unless it completes within an implementation-specific, bounded time limit._

So what really should be the behavior of WFI in User mode, when mstatus=TW=0? And is the last statement above should be read to imply that it can never truly "wait indefinitely" for an interrupt .. it must be guaranteed to complete.

scottj97, 2022-08-02:
I'm not sure I understand the contradiction you see? In Spike, the implementation-specific time limit is 0.

aprnath, 2022-08-02:
If the time limit is 0, shouldn't it always succeed?

aswaterman, 2022-08-02:
Other way around.  If the time limit is 0, then it is impossible for WFI to complete within an implementation-specific, bounded time limit; hence, it'll raise the exception.

aprnath, 2022-08-03:
Got it. Thank you.

closed 2022-08-03 by aprnath

algrobman, 2022-08-03:
Isn't all this mean, that the spec requires better wording of the WFI chapter. Our folks go back and forth trying to determine what's right behavior for this instruction for a week already. Every time somebody brings an argument for opposite behavior, referring to different spec sentence . There are following outcomes for WFI execution: 
a) wait for interrupt state  
b) illegal trap
c) Noop
A table for these outcomes conditions is highly appreciated.

scottj97, 2022-08-08:
One question is when WFI is allowed to wait forever. (It's always okay for it not to wait, or to wait no more than an implementation-specific, bounded time limit.) I think the spec is clear with all the statements about "implementation-specific, bounded time limits."

Another question: when it's not allowed to wait forever, what must happen when no interrupt is pending? It seems it can either be a NOP (possibly a very slow NOP), or take an illegal instruction exception. I don't see any requirement in the spec preferring one of those over the other.

It seems that Spike takes illegal/virtual instruction exceptions without regard to any pending interrupts, even when it would be legal to do NOP in that case. Apparently it is legal to take those exceptions even when the WFI instruction would complete within an implementation-specific, bounded time limit.

algrobman, 2022-08-09:
Does the "implementation-specific, bounded time limit." mean that even in the M-mode and when mstatus.TW=0 and mie (csr) = 0 WFI should not stop the CPU, but trap as illegal or be executed as NOP? (i.e. should CPU logic  detect conditions, where it can't resume execution after WFI  and execute WFI accordingly (?)).  

On other hand, the CPU still can get stuck indefinitely after WFI, if mie is not zero, but platform devices are not programmed  to generate  interrupt(s)

scottj97, 2022-08-09:
> Does the "implementation-specific, bounded time limit." mean that even in the M-mode and when mstatus.TW=0 and mie (csr) = 0 WFI should not stop the CPU, but trap as illegal or be executed as NOP?

NOP is always okay, but the spec has no provision for an exception under the conditions where an unlimited wait is allowed, such as what you describe here.

> (i.e. should CPU logic detect conditions, where it can't resume execution after WFI and execute WFI accordingly (?)).

You mean detect that MIE=0, so it can never awake, and react? I see no reason to do this, but since NOP is always allowed, it's probably not a spec violation to do so. But it can be useful for software to do such a thing...for example a single-threaded application in a dual-hart system could do WFI on the unused hart and never need it to awake.

> On other hand, the CPU still can get stuck indefinitely after WFI, if mie is not zero, but platform devices are not programmed to generate interrupt(s)

Yes. If the OS screws up, WFI can hang the hart until hard reset (or NMI, if implemented).