riscv-software-src/riscv-isa-sim #1060: Which Debug specification version is targeted?

YenHaoChen, 2022-08-04:
Hello,

There are two pre-built PDFs on the top of the official riscv-debug-spec website (https://github.com/riscv/riscv-debug-spec).
One is the "latest stable" Version 1.0.0-STABLE (Jun 9 2022), and the other one is the "latest release" Version 0.13.2 (Mar 22 2019).
The Debug 0.14 on the spike website does not match up with any of these.

What is the debug specification version the spike is targeting?

Best regards,
YenHaoChen

aswaterman, 2022-08-04:
@timsifive

YenHaoChen, 2022-08-09:
@aswaterman @timsifive 
Version 1.0.0-stable states, "Version 0.14 was a working version that was never officially ratified." I guess that means Version 0.14 is outdated.

I need the Sdext extension (debug mode extension). Will spike accept PRs for debug Version 1.0.0-stable?

timsifive, 2022-08-18:
Spike aims to be compatible with debug spec 1.0. (1.0 isn't final yet, but backwards incompatible changes are no longer accepted.)

YenHaoChen, 2022-08-18:
Thank you for the reply.

closed 2022-08-18 by YenHaoChen