riscv-software-src/riscv-isa-sim #1064: Dead code in VI_VV_EXT macro

ksco, 2022-08-09:
In the VI_VV_EXT macro, the pat variable is constructed such that the upper 4 bits represent the target element width and the lower 4 bits represent the source element width, using the following mapping:

- 1 -> 8
- 2 -> 16
- 4 -> 32
- 8 -> 64

For example, 0x81 represents SEW = 64 and EEW = 8. But at [Line 1471](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/v_ext_macros.h#L1471), pat is expected to be equal to 0x88. According to the spec, this seems never to happen in a legal instruction.

> riscv-v-spec-1.0.pdf
> 11.3. vector integer extension
> "The vector integer extension instructions zero- or sign-extend a source vector integer operand with **EEW less than SEW** to fill SEW-sized elements in the destination."

scottj97, 2022-08-09:
It's also interesting (and curious) how the code does the same thing for 0x88 as it does for 0x84.

aswaterman, 2022-08-09:
Indeed, the destination EEW is always strictly larger than the source EEW for vsext/vzext, and so I don't know what the 0x88 case is for.  I assume it can be safely deleted (cc @chihminchao)

ksco, 2022-08-10:
Thanks for confirming this, I have submitted a PR.

PR #1065 (2022-08-11): Remove dead code in VI_VV_EXT macro
Fix #1064 

Tested with the following files, but the test coverage may not be enough though.

[vzext.vf2_LMUL2SEW16.S](https://github.com/ksco/riscv-tests/blob/spike/isa/rv64uv/vzext.vf2_LMUL2SEW16.S)
[vzext.vf2_LMUL2SEW32.S](https://github.com/ksco/riscv-tests/blob/spike/isa/rv64uv/vzext.vf2_LMUL2SEW32.S)
[vzext.vf2_LMUL2SEW64.S](https://github.com/ksco/riscv-tests/blob/spike/isa/rv64uv/vzext.vf2_LMUL2SEW64.S)
[vzext.vf2_LMUL4SEW16.S](https://github.com/ksco/riscv-tests/blob/spike/isa/rv64uv/vzext.vf2_LMUL4SEW16.S)
[vzext.vf2_LMUL4SEW32.S](https://github.co [... cut]
closed 2022-08-11 by aswaterman