riscv-software-src/riscv-isa-sim #1077: Should zext32 be used in fcvt_wu_d?

soberl, 2022-09-03:
https://github.com/riscv-software-src/riscv-isa-sim/blob/476a79fe2fc6c4c7b809d0b359ad041a1034ffce/riscv/insns/fcvt_wu_d.h#L4

#define sext32(x) ((sreg_t)(int32_t)(x))
#define zext32(x) ((reg_t)(uint32_t)(x))

Here if the result of conversion >= 0x80000000, the sext32 and zext32 give different result. Not sure what's the target value in RV64?

aswaterman, 2022-09-03:
It's correct as-is: https://github.com/riscv/riscv-isa-manual/blob/2713791c03c8943656dc6f71a39c5dd0916f5260/src/f.tex#L563

32-bit integers are generally held sign-extended in x-registers on RV64, even when they represent unsigned quantities.

closed 2022-09-03 by aswaterman

soberl, 2022-09-03:
Thanks for the explanation. 
Yes, that likes the FP16 in 32bits FPR.