riscv-software-src/riscv-isa-sim #1078: Is 'require_vector_novtype'  correct for  vmvnfr.v instruction now ?

xiongtao2022, 2022-09-06:
I have seen  that vmvnfr_v.h  was fixed to take vsew into account,  but it still use 'require_vector_novtype' function , which won't raise a trap if vill is set.  As spec said, "If the vill bit is set, then any attempt to execute a vector instruction that depends upon vtype will raise an illegal-instruction exception". 
Also, "These instructions are intended to aid compilers to shuffle vector registers without needing to know or change vl or vtype." seems contradictory now.

aswaterman, 2022-09-06:
Agreed this is a bug.  See https://github.com/riscv-software-src/riscv-isa-sim/pull/1079

I actually disagree with your comment that the spec sentence is contradictory, as it can be assumed that in correct operation, `vill`=0.  The important thing is that `vmv<nf>r.v` has the same architecturally visible effect regardless of SEW when `vill`=0.

xiongtao2022, 2022-09-07:
Sorry, I didn't describe last sentence clearly. I mean that vmv<nf>r.v needs to know vsew in vtype when vstart !=0, but spec said    "without needing to know vtype", that confuse me.

PR #1079 (2022-09-07): vmv<nf>r.v depends on vtype, and therefore should check vill
Resolves #1078
closed 2022-09-07 by aswaterman