riscv-software-src/riscv-isa-sim #1082: Whether mstatus.VS should be set to dirty when the vector instructions trigger illegal instruction exception?

liweiwei90, 2022-09-09:


aswaterman, 2022-09-09:
If VS = 0, that would be illegal. If VS != 0, it is legal but not preferable.

liweiwei90, 2022-09-09:
I know this check which is done by require_vector_vs. However, the vector instructions have many other checks. What I really means is whether dirty_vs_state should be called when the other check fails.

aswaterman, 2022-09-09:
The best option is to avoid dirtying the state if an illegal-instruction exception is raised. But there are some cases where we can make a choice.

liweiwei90, 2022-09-09:
I find require_vector is often checked before other checks in many cases. So dirtying the state may usually happen when he other check fails.

aswaterman, 2022-09-09:
That’s technically OK, as long as the state is never dirtied when VS=0. We could choose to make this more precise. Perhaps others should weigh in.

liweiwei90, 2022-09-09:
OK.

scottj97, 2022-09-09:
If it's simply a matter of reordering the checks before any state is updated, I'd support that improvement.

aswaterman, 2023-04-25:
@mlawson-tt Actually, I think you identified a bug.  Because `fcvt.l.s` writes the FP exception flags, it must mark FS dirty (at least if the flags are actually changed).

Similarly, many vector instructions that don't explicitly write vector registers must still set VS to dirty (because of the saturation flag, for example).

mlawson-tt, 2023-04-25:
@aswaterman Ah, I wasn't clear enough there.  `fcvt.l.s` does mark FS as dirty if it produces any exception flags (even if they don't change), but it does not otherwise (whereas `fcvt.s.l` marks FS as dirty regardless as long as it is not illegal)  Spike also marks FS as dirty if on any of the writes to FP CSRs through the `float_csr_t::unlogged_write`, and the same occurs for VS through `vector_csr_t::unlogged_write` and `vxsat_csr_t::unlogged_write`, so I think the saturation case occurs.

aswaterman, 2023-04-25:
Right, thanks for the clarification.  Anyway, to answer your question, yes, we have the option of not setting VS to dirty for some vector instructions (e.g. `vmv.x.s`, or a vector store that is executed with `vstart`=0 and does not trap, thus keeping `vstart` unchanged).  Not sure it's worth the effort, but it is an option.

Note that, as implemented, vector instructions explicitly write `vstart` to 0 after completing.  For the cases we'd be trying to filter, that would need to become something like `if (P.VU.vstart->read() != 0) P.VU.vstart->write(0);` to avoid dirtying VS when `vstart` doesn't actually change.

mlawson-tt, 2023-04-25:
Currently, FS is set dirty if the FP register is written to at all even if the value doesn't change (including the flags), so I think VS could work the same way.  As a result, the `vstart` case would be OK as is because `vstart` is being written to even though the value doesn't change.

Really I think the strongest desire here would just be for FS and VS to behave consistently.  It doesn't really matter which method they use (though I tend to think the current FS behavior of marking as dirty if a register is written to even if the value doesn't change makes sense), but ideally they just both do the same thing in the same scenarios (or at least as close as they can get).  It's a pretty minor issue for the most part, so yeah I'm not sure it's really necessarily worth changing if the current behavior is intentional.  I'm mostly just curious if there is a specific philosophy for when these FS/VS/etc. get marked dirty outside the general requirements in the spec.

Note I haven't looked at it super closely, but it's a possibility that everything works as desired already or is very close because if the instruction is already writing `vstart` at the end, VS will be dirty anyway based on the mark-as-dirty-if-writing-anything behavior.

aswaterman, 2023-04-25:
@mlawson-tt I agree both with the goal to make FS and VS behave similarly, and with the observation this is almost always the case because of the `vstart` write.

mlawson-tt, 2023-04-25:
@aswaterman awesome, thanks for the info!