riscv-software-src/riscv-isa-sim #1087: Fix some compiler warnings?

scottj97, 2022-09-22:
This nice-looking project [Hammer](https://github.com/rivosinc/hammer) requires a patch to Spike to work. Some of it is essential (building Spike as a library instead of standalone executable) but much of [their patch](https://github.com/rivosinc/hammer/blob/2a328c63fd565d0cdfc897bbdd0179bcd0aec980/spike.ccc97918.hammer.patch) seems to be disabling various compiler warnings.

Should we fix these issues so the warnings don't need to be disabled?

cc @jerinjoy

aswaterman, 2022-09-22:
Yes, better to compile cleanly. I can take a stab, unless you are already on it.

scottj97, 2022-09-22:
> Yes, better to compile cleanly. I can take a stab, unless you are already on it.

No, it's possible these only apply in clang, which I don't have ready access to.

aswaterman, 2022-09-22:
🍁

jerinjoy, 2022-09-22:
Thanks for the interest in Hammer! If neither of you has started on fixing the warnings I can work on this. Let me know.

aswaterman, 2022-09-22:
@jerinjoy I've fixed several already and will make a PR shortly when I've run out of steam.  (But, for a handful of them, the "fix" is unnecessary and clutters the code, and so the pragmas probably really are the way to go.)

aswaterman, 2022-09-23:
OK, officially out of steam.  I fixed all of the ones that GCC understands.  https://github.com/riscv-software-src/riscv-isa-sim/pull/1089

@jerinjoy Can you handle the rest?  You can look at that PR to get a rough idea for when I thought it was best to give up and use the pragmas.

jerinjoy, 2022-09-23:
There was one additional warning `-Wnon-virtual-dtor` that was causing the hammer build to fail. I've fixed this in this PR:
https://github.com/riscv-software-src/riscv-isa-sim/pull/1091

With these changes, Hammer will build without errors. We still need to patch in the Makefile changes to build the Spike shared libraries. Do we also want to pull these changes into Spike?

aswaterman, 2022-09-23:
Thanks for the help! I’ll place the ball back in @scottj97’s court.

scottj97, 2022-09-23:
> We still need to patch in the Makefile changes to build the Spike shared libraries. Do we also want to pull these changes into Spike?

I'd be okay with that (as long as the standalone executables still build and work, of course) but that should be a separate PR.

jerinjoy, 2022-09-23:
New PR with the Makefile changes to build the Spike shared libraries:
https://github.com/riscv-software-src/riscv-isa-sim/pull/1093

scottj97, 2022-11-08:
Looks like this was all completed as of #1093.

closed 2022-11-08 by scottj97