riscv-software-src/riscv-isa-sim #1090: vestvli - unrecognized opcode vsetvli a5,a5,e32

balaji-ch, 2022-09-23:
Hi
I tried the following code 
```c
#include <stdio.h>

void main(){

    //unsigned long vlenb;
    //asm volatile ("csrr %0, vlenb" : "=r" (vlenb));

    unsigned long  vl = 0;
    unsigned long  elements = 128;
    asm volatile ("vsetvli %0, %1, e32" : "=r"(vl) : "r"(elements));
    printf("vl init = %d\n",vl);

}
```
with the below build command
` riscv64-unknown-elf-gcc -o hello hello.c`

I get the following error
`hello.c:9: Error: unrecognized opcode `vsetvli a5,a5,e32'`

What is the correct way of using vsetvli ?

balaji-ch, 2022-09-23:
Figured out the correct way

**Build**
```sh
riscv64-unknown-elf-gcc -march=rv64gcv -o hello hello.c
```

**Run**
```sh
spike --varch=vlen:128,elen:32 --isa=rv64gcv pk hello
```

aswaterman, 2022-09-23:
Indeed, the error you encountered has nothing to do with Spike.

closed 2022-09-23 by aswaterman

balaji-ch, 2022-09-24:
It would be better if this is added to the README.md of Spike RISC-V ISA Simulator