riscv-software-src/riscv-isa-sim #1097: Fix RVV vill check in set_vl()

jerinjoy, 2022-09-23:
vill was being incorrectly set to true by this condition:

vsew > std::min(vflmul, 1.0f) * ELEN
The LMUL tells us what fraction of the vector register will be used so VLEN (vector register size) should be used instead of ELEN (maximum element size).

Fix here:
https://github.com/riscv-software-src/riscv-isa-sim/pull/1095

aswaterman, 2022-09-23:
Not a bug, as I mentioned on the ticket.

jerinjoy, 2022-09-23:
Thanks for the clarification. Closing.

jerinjoy, 2022-09-25:
Reopening to clarify.

As you mentioned, the spec says:

```
the requirement is to support LMUL ≥ SEWMIN/ELEN, where SEWMIN is the narrowest supported SEW value and ELEN is the widest supported SEW value.
```

Setting VILL if  `vsew > std::min(vflmul, 1.0f) * ELEN` doesn't seem to match this though as we're not taking SEWMIN into consideration here. Shouldn't VILL be set if `LMUL < SEWMIN/ELEN` instead?

jerinjoy, 2022-09-26:
@aswaterman 

An example to better explain this:

For this instruction, VL should be set to (VLEN * 0.5)/64:

```
vsetvli a1, zero, e64, mf2
```

This is the condition that sets `VILL`:

```
vsew > std::min(vflmul, 1.0f) * ELEN
```

In this case `vsew` is `64` and `vflmul` is `0.5`. If `ELEN` is `64` then this condition becomes true and the `VILL` will be set.

If we want to enforce this requirement:

```
the requirement is to support LMUL ≥ SEWMIN/ELEN, where SEWMIN is the narrowest supported SEW value and ELEN is the widest supported SEW value.
```

as a check the condition should be:

```
((float)vsewmin/ELEN) > vflmul
```

Is vsewmin implementation defined?

aswaterman, 2022-09-26:
Sorry, I quoted the wrong statement in the spec to justify my point.  The relevant statement is: "For a given supported fractional LMUL setting, implementations must support SEW settings between SEWMIN and LMUL * ELEN, inclusive"

So, for LMUL = 1/2, implementations must support SEW settings between 8 and 32.  Implementations are allowed to support greater SEW settings (provided they fit within VLEN * LMUL), but are not required to.

So, Spike is implementing the conservative bound, and you are proposing changing it to the liberal bound.  Both are correct.  The reason I am pushing back is the conservative bound catches software portability errors.  Software that uses SEW=64 LMUL=1/2 will raise illegal-instruction traps on some correct implementations but not others.  Better to catch that as early as possible.

closed 2022-10-10 by jerinjoy

YenHaoChen, 2022-10-26:
> The relevant statement is: "For a given supported fractional LMUL setting, implementations must support SEW settings between SEWMIN and LMUL * ELEN, inclusive"

@aswaterman 
I cannot find the sentence in riscv-v-spec-1.0 (https://github.com/riscv/riscv-v-spec/releases/tag/v1.0). Can you point out the page of the sentence?

aswaterman, 2022-10-26:
I just used Ctrl+F and found it immediately... it is at the bottom of page 11.

YenHaoChen, 2022-10-27:
Thank you. I found it with your index. 
(It seems my software searching engine has an issue with the file.)
[image]