riscv-software-src/riscv-isa-sim #1101: libriscv.a not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64

grejioh, 2022-09-28:
Using "brew install riscv-tools". Is this because of Xcode 14.0?
os:  macOS 12.6
MacBook Air M2
```
==> Installing riscv/riscv/riscv-tools dependency: riscv-isa-sim
==> ../configure --prefix=/opt/homebrew/Cellar/riscv-isa-sim/main --with-target=../../riscv-pk/main/riscv64-unknown-elf --without-boost --without-boost-asio --without-boost-regex
==> make
Last 15 lines from /Users/grejioh/Library/Logs/Homebrew/riscv-isa-sim/02.make:
make: *** Waiting for unfinished jobs....
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 12316840 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 12316840 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
make: *** [termios-xspike] Error 1
make: *** [xspike] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [spike] Error 1
ld: in libspike_main.a(libriscv.a), archive member 'libriscv.a' with length 12316840 is not mach-o or llvm bitcode file 'libspike_main.a' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [libcustomext.so] Error 1
ld: in libspike_dasm.a(libriscv.a), archive member 'libriscv.a' with length 12316840 is not mach-o or llvm bitcode file 'libspike_dasm.a' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [spike-dasm] Error 1
```

grejioh, 2022-09-28:
Maybe because of this:
[Xcode-14-release-notes](https://developer.apple.com/documentation/xcode-release-notes/xcode-14-release-notes)

> Deprecations
Starting with Xcode 14, bitcode is no longer required for watchOS and tvOS applications, and the App Store no longer accepts bitcode submissions from Xcode 14.
Xcode no longer builds bitcode by default and generates a warning message if a project explicitly enables bitcode: “Building with bitcode is deprecated. Please update your project and/or target settings to disable bitcode.” The capability to build with bitcode will be removed in a future Xcode release. IPAs that contain bitcode will have the bitcode stripped before being submitted to the App Store. Debug symbols for past bitcode submissions remain available for download. (86118779)

aswaterman, 2022-09-28:
This actually looks like a bug in the Xcode toolchain.  Other people say it results from attempting to link debug and non-debug objects, which is supposed to work correctly.  I'm trying to find a workaround.

aswaterman, 2022-09-28:
Nevermind, it's not an Xcode problem.  Fixed in #1104.

closed 2022-09-28 by aswaterman

grejioh, 2022-09-28:
> Nevermind, it's not an Xcode problem. A new sanity check catches something weird (if not technically incorrect) that we were doing. Fixed in #1104.

Everything is good! Thank you very much!🫰

PR #1104 (2022-09-28): Fix build using Xcode 14
Don't include .a files in other .a files.  Doing so trips a sanity check in the Xcode 14 toolchain, because the included .a files are not themselves mach-o format, even though their contents are.

Fixes #1101