riscv-software-src/riscv-isa-sim #1106: illegal error on the master branch but does not exist on the v1.1.0 tag

lhtin, 2022-09-30:
Hi, I got an illegal error on the master branch, the illegal instruction is `vmv1r.v v25,v8`. And I can't reproduce the error on the v1.1.0 tag. Thanks.

error message:

```
bbl loader
z  0000000000000000 ra 000000000001015c sp 0000003ffffffb50 gp 0000000000011c80
tp 0000000000000000 t0 0000000000010244 t1 000000000000000f t2 0000000000000000
s0 0000000000000000 s1 0000000000000000 a0 0000000000000001 a1 0000003ffffffb58
a2 0000000000000000 a3 0000000000000010 a4 0000000000000001 a5 0000000000000000
a6 000000000000001f a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000010106 va/inst 000000009e803cd7 sr 8000000200006620
An illegal instruction was executed!
```

Reproduce C code:

```c
int main()
{
        asm volatile ("vmv1r.v v25,v8");
}
```

aswaterman, 2022-09-30:
This probably isn't a bug.  I don't see any evidence that `vsetvli` has been executed at any point in this program, so `vtype.vill` might be set.  (`vmv<nf>r.v` instructions depend on `vtype` and hence require that `vtype.vill`=0.)

lhtin, 2022-09-30:
We didn't consider the dependence of `vtype` when developing GCC, and I try it with LLVM compiler and LLVM also doesn't emit `vsetvli` for `vmv<nf>r.v`([here](https://godbolt.org/z/xEPdTG33T) the online result you can reproduce).

Can I ask why the `vmv<nf>r.v` instruction need the `vtype` to be set? And Is the `vs<nf>r.v` also need the `vtype`?

aswaterman, 2022-09-30:
@lhtin I don't think the compiler needs to emit any additional instructions.  The convention should be that `vtype.vill` is 0 upon entry.  Then, it is the responsibility of `crt0` (or the OS kernel, it doesn't really matter which) to make sure `vtype.vill` is 0 before calling into any compiled code.  This can be accomplished by executing a single `vsetivli x0, 0, e8, m1, ta, ma` instruction.

This isn't the right forum to debate the choices in the ISA spec, and in any case, the behavior is already ratified.  The short answer is that the instruction's behavior is SEW-dependent because of interrupts: `vstart` is in increments of elements, and so the width of the element matters.

aswaterman, 2022-09-30:
I am closing this issue because it is not a Spike bug.

closed 2022-09-30 by aswaterman