riscv-software-src/riscv-isa-sim #1108: Redeclaration in encoding.h and debug_defines.h

YenHaoChen, 2022-10-07:
There are redeclarations of CSR fields in encoding.h and debug_defines.h. For instance, the mcontrol.select is declared by both encoding.h and debug_defines.h but with different names, MCONTROL_SELECT and CSR_MCONTROL_SELECT. Thus, there are two naming conventions for a CSR field.

The current source code uses both naming conventions, e.g., CSR_MCONTROL_HIT and MCONTROL_SELECT. Do we prefer to use a specific naming convention for consistency? Or do we accept both of the naming conventions?

aswaterman, 2022-10-07:
encoding.h is an auto-generated source of truth, so we should generally use its definitions and avoid manually duplicating them into other files.

YenHaoChen, 2022-10-10:
Thank you, @timsifive, for the further comment.

closed 2022-10-13 by YenHaoChen