riscv-software-src/riscv-isa-sim #1110: Issues printing commit logs

rtorre32, 2022-10-10:
I did the ../configure --enable-commitlog --prefix=$RISCV
When I run my code using , spike --log-commits pk a.out
I get the error -- Commit logging support has not been properly enabled; please re-build the riscv-isa-sim project using "configure --enable-commitlog".

What Am I doing wrong?

scottj97, 2022-10-10:
Are you doing an incremental build or a clean build? Incremental probably won't rebuild enough. Try deleting your build directory first, so you get a clean build.

rtorre32, 2022-10-10:
> Are you doing an incremental build or a clean build? Incremental probably won't rebuild enough. Try deleting your build directory first, so you get a clean build.

I don't know If I am missing a command, but I deleted build directory created it again, went in it and ran the command ../configure --enable-commitlog --prefix=$RISCV and that is it. I then ran my a.out with the same command as above and I get the same error.

scottj97, 2022-10-10:
> I don't know If I am missing a command, but I deleted build directory created it again, went in it and ran the command ../configure --enable-commitlog --prefix=$RISCV and that is it. I then ran my a.out with the same command as above and I get the same error.

I assume you ran `make` too? After `configure`? Are you sure you're running `./spike` and not picking up another spike somewhere in your $PATH?

rtorre32, 2022-10-11:
> I assume you ran `make` too? After `configure`? Are you sure you're running `./spike` and not picking up another spike somewhere in your $PATH?

I completely deleted my spike, when I run spike it says command not found. I cloned the repo once again, went in it made build directory, went inside build and ran the commands 
../configure --enable-commitlog --prefix=$RISCV
../configure --prefix=$RISCV
make

in that order exactly, now when I try to run my code using spike pk a.out it says spike command not found.
It might be an issue of it not being on my path, what should I add to my path so that the spike command works?

scottj97, 2022-10-11:
Until you run `make install` it only lives in the current directory. So you'd want to run `./spike`

(Note: not a good idea to add `.` to your $PATH.)

scottj97, 2022-11-08:
@rtorre32 can we close this as solved?

rtorre32, 2023-01-26:
> @rtorre32 can we close this as solved?

Yes, Thanks!

closed 2023-01-26 by aswaterman