riscv-software-src/riscv-isa-sim #1127: How to use spike to run a fedora distribution？

yorange1, 2022-10-24:
Hello everyone，I'm trying to run an image of a fedora's riscv distribution using spike，I have made a bootloader using Opensbi and uboot, but I don't know how to mount the fedora image file into the system for uboot to boot. 
I see that there is a +disk option in the htif interface in spike, but it doesn't seem to support it now. I would like to hear everyone's advice, thank you very much.

yorange1, 2022-10-24:
My fedora image file is downloaded from here，https://dl.fedoraproject.org/pub/alt/risc-v/repo/virt-builder-images/images/

yorange1, 2022-10-26:
Or, are there plans for spike to support virtio interfaces like qemu?

aswaterman, 2022-10-26:
Spike isn't a full-system simulator.  Although it can boot a Linux kernel, it can't support a full distro.  We don't have plans to make Spike more elaborate at this time, so I'd recommend using qemu.

yorange1, 2022-11-01:
> Spike isn't a full-system simulator. Although it can boot a Linux kernel, it can't support a full distro. We don't have plans to make Spike more elaborate at this time, so I'd recommend using qemu.

thanks for your advice.

closed 2022-11-01 by yorange1