riscv-software-src/riscv-isa-sim #1140: User load segfault @ 0x40a636a8bb8f62dc  after running spike pk coremark.exe

dshm, 2022-11-03:
I am running coremark test using spike pk but get the User load segfault @ 0x40a636a8bb8f62dc, I can't find a way to solve this ,could somebody give me some help?
the toolchain I use is riscv64-unknown-elf-gcc, I install it using crosstool-NG

the commands are as below:
export RISCV=/data/dushaomin/issuetest/installtest
export PATH=/home/dushaomin/x-tools/riscv64-unknown-elf/bin:$PATH
git clone https://github.com/riscv-software-src/riscv-isa-sim.git
using the code in master branch, commit id is ab3225a3ff687fda8b4180f9e4e0949a400d1247

mkdir build;cd build
../configure --prefix=$RISCV
make
make install

git clone https://github.com/riscv-software-src/riscv-pk.git
using the code in master branch, commit id is 7e9b671c0415dfd7b562ac934feb9380075d4aa2
mkdir build; cd build
../configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gc_zifencei
make
make install

add pk and spike to PATH
export PATH=/data/dushaomin/issuetest/installtest/bin:$PATH
export PATH=/data/dushaomin/issuetest/installtest/riscv64-unknown-elf/bin:$PATH
the hello world test is ok running spike pk hello as in README.

git clone https://github.com/eembc/coremark
use the coremark code in master branch, commit id 
eefc986ebd3452d6adde22eafaff3e5c859f29e4
cp -r simple riscv64

modify the core_portme.mak in riscv64 to chang the compiler as riscv64-unknown-elf-gcc
line 24 CC = riscv64-unknown-elf-gcc
line 45 RUN = spike pk

make PORT_DIR=riscv64
and get the error in ruin1.log

bbl loader^M
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 14068014
Total time (secs): f
Iterations/Sec   : f
Iterations       : 40000
Compiler version : GCC11.2.0
Compiler flags   : -O2 -DPERFORMANCE_RUN=1
Memory location  : STACK
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0x25b5
Correct operation validated. See README.md for run and reporting rules.
z  0000000000000000 ra 000000000001257c sp 0000003ffffff050 gp 00000000000173a8
tp 0000000000000000 t0 0000000000000019 t1 0000000000000009 t2 000000000000002a
s0 0000000000016f88 s1 00000000000169a0 a0 40a636a8bb8f62dc a1 0000000000000000
a2 ffffffffffffffff a3 0000000000000001 a4 0000003ffffff0e0 a5 40a636a8bb8f62dc
a6 0000000000012370 a7 000000000000000a s2 0000003ffffff0d0 s3 0000003ffffff1e8
s4 0000000000000000 s5 000000000001537e s6 40a636a8bb8f62dc s7 ffffffffffffffff
s8 0000003ffffff1f0 s9 0000002400000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000078 t4 000000000000004e t5 0000000000000039 t6 000000000000001f
pc 0000000000011f7a va/inst 40a636a8bb8f62dc sr 8000000200006020
User load segfault @ 0x40a636a8bb8f62dc

I also tried the https://github.com/riscv-boom/riscv-coremark.git for the same test and get the same error. I have tried above process on two different machine and get the same error. I am not sure where is wrong. Coule someboby please give me some adivce?

releated isseue
https://github.com/riscv-software-src/riscv-pk/issues/285
https://github.com/riscv-boom/riscv-coremark/issues/11

aswaterman, 2022-11-03:
We don’t have any reason to believe this is a bug in Spike. You should figure out what’s going wrong and report back for posterity.

closed 2022-11-03 by aswaterman

dshm, 2022-11-09:
the reason is in core_main.c ee_printf error when printf float type varibale(line 409). the riscv64-unknown-elf-gcc installed using crosstool-NG is not support float type IO by default. changing the .config file generated by crosstool-NG and set CT_LIBC_NEWLIB_IO_FLOAT=y, re-build and re-install the  riscv64-unknown-elf-gcc will solve the problem

aswaterman, 2022-11-09:
Thank you for reporting back. Hopefully others will find your conclusion helpful.