riscv-software-src/riscv-isa-sim #1142: commit log

rtorre32, 2022-11-07:
I have been trying to figure out what the log commits are printing but I can't seem to find what it means. 

core   0: 3 0x0000000000001000 (0x00000297) x5  0x0000000000001000
core   0: 3 0x0000000000001004 (0x02028593) x11 0x0000000000001020
core   0: 3 0x0000000000001008 (0xf1402573) x10 0x0000000000000000
core   0: 3 0x000000000000100c (0x0182b283) x5  0x0000000080000000 mem 0x0000000000001018
core   0: 3 0x0000000000001010 (0x00028067)

This is an example of a commit log printed.

rtorre32, 2022-11-07:
> Your example looks incomplete to me because it doesn't include any instruction disassembly lines?
> 
> This is what I see:
> 
> ```
> core   0: 0x00000000800022e2 (0xfa62d2e3) bge     t0, t1, pc - 92
> core   0: 3 0x00000000800022e2 (0xfa62d2e3)
> core   0: 0x00000000800022e6 (0x0002b203) ld      tp, 0(t0)
> core   0: 3 0x00000000800022e6 (0x0002b203) x4  0x0000000080002380 mem 0x0000000080002428
> core   0: 0x00000000800022ea (0x00008202) c.jr    tp
> core   0: 3 0x00000000800022ea (0x8202)
> ```
> 
> The first line above is PC, opcode, and disassembly of the executed instruction. The second line is the effects of that instruction: register updates, memory reads & writes.
> 
> I'm not sure what the `3` near the beginning of the second line means.

Would you have an idea of why my logs are not showing any instruction disassembly lines? I do get this error at the top of my log file
<stdin>:48.24-56.7: Warning (interrupt_provider): /soc/plic@c000000: Missing #address-cells in interrupt provider
I'm not sure exactly what it means.

All I did was enable log commits when I built the spike, is there anything else I need to enable?

scottj97, 2022-11-07:
> Would you have an idea of why my logs are not showing any instruction disassembly lines?

No idea. That is weird.

omerguzelelectronicguy, 2022-11-09:
@rtorre32 could you share with which command you ran spike? 
I remember I faced a similar issue, not sure. However, I couldn't remember how I faced and solved it.

rtorre32, 2022-11-16:
> @rtorre32 could you share with which command you ran spike? I remember I faced a similar issue, not sure. However, I couldn't remember how I faced and solved it.

I ran it with 

spike --log-commits pk a.out 2>"FileName"

where 'FileName' is the name of the file where the log is being printed

omerguzelelectronicguy, 2022-11-17:
ok. I think you should use it with `-l` flag.
For example, those are two different examples that I used Spike

example with `--log-commits`
```
/opt/rv32gcv/bin/spike  --log-commits  --isa=rv32gcv --varch=vlen:128,elen:32 ./TestCode/temporary_files/hi_rv.elf
core   0: 3 0x00001000 (0x00000297) x5  0x00001000
core   0: 3 0x00001004 (0x02028593) x11 0x00001020
core   0: 3 0x00001008 (0xf1402573) x10 0x00000000
core   0: 3 0x0000100c (0x0182a283) x5  0x80000000 mem 0x00001018
core   0: 3 0x00001010 (0x00028067)
core   0: 3 0x80000000 (0x4081) x1  0x00000000
core   0: 3 0x80000002 (0x4101) x2  0x00000000
core   0: 3 0x80000004 (0x4181) x3  0x00000000
core   0: 3 0x80000006 (0x4201) x4  0x00000000
core   0: 3 0x80000008 (0x4281) x5  0x00000000
core   0: 3 0x8000000a (0x4301) x6  0x00000000
```

example with `-l --log-commits`
```
/opt/rv32gcv/bin/spike -l --log-commits  --isa=rv32gcv --varch=vlen:128,elen:32 ./TestCode/temporary_files/hi_rv.elf
core   0: 0x00001000 (0x00000297) auipc   t0, 0x0
core   0: 3 0x00001000 (0x00000297) x5  0x00001000
core   0: 0x00001004 (0x02028593) addi    a1, t0, 32
core   0: 3 0x00001004 (0x02028593) x11 0x00001020
core   0: 0x00001008 (0xf1402573) csrr    a0, mhartid
core   0: 3 0x00001008 (0xf1402573) x10 0x00000000
core   0: 0x0000100c (0x0182a283) lw      t0, 24(t0)
core   0: 3 0x0000100c (0x0182a283) x5  0x80000000 mem 0x00001018
core   0: 0x00001010 (0x00028067) jr      t0
core   0: 3 0x00001010 (0x00028067)
core   0: 0x80000000 (0x00004081) c.li    ra, 0
core   0: 3 0x80000000 (0x4081) x1  0x00000000
core   0: 0x80000002 (0x00004101) c.li    sp, 0
core   0: 3 0x80000002 (0x4101) x2  0x00000000
core   0: 0x80000004 (0x00004181) c.li    gp, 0
core   0: 3 0x80000004 (0x4181) x3  0x00000000
core   0: 0x80000006 (0x00004201) c.li    tp, 0
core   0: 3 0x80000006 (0x4201) x4  0x00000000
core   0: 0x80000008 (0x00004281) c.li    t0, 0
core   0: 3 0x80000008 (0x4281) x5  0x00000000
core   0: 0x8000000a (0x00004301) c.li    t1, 0
core   0: 3 0x8000000a (0x4301) x6  0x00000000
core   0: 0x8000000c (0x00004381) c.li    t2, 0
```

In the second example including `-l --log-commits`, as I know, the first line shows core <core_id> , the address of the instruction, hex representation of the instruction, and assembly instruction respectively.
The second line is similar to the previous line but instead of the assembly instruction, there is a changed thing (can be memory GPR, FPR, VPR, or CSR) and a new value of the thing. 
I don't know too the `3` in the first line.

I wonder if there is a flag for Spike that we can see all the things at the same line instead of each cycle 2 line.

omerguzelelectronicguy, 2022-11-17:
When you gave only `--log-commits` but with a `-d` flag you will again get a similar result with `-l --log-commits` but of course in debug mode.

jerryz123, 2022-12-05:
I prefer retaining the ability to generate a log w/o disassembly, since it minimizes the size of the log when piped to a file.

omerguzelelectronicguy, 2022-12-05:
I prefer retaining too. Even there should be a way to minimize the log more with interactive debug commands such as pc jumping 
 (`until pc <core> <val>`) I think.

rtorre32, 2022-12-06:
> ok. I think you should use it with `-l` flag. For example, those are two different examples that I used Spike
>
> example with `--log-commits`
>
> ```
> /opt/rv32gcv/bin/spike  --log-commits  --isa=rv32gcv [...]

Thank you, this worked.