riscv-software-src/riscv-isa-sim #1148: Starting Spike halted breaks GDB interface

thatliuser, 2022-11-19:
Hello,

I am currently running into issues trying to debug a program running under pk, but only when Spike is started halted.
When Spike is started normally with `--rbb-port` with a wait loop in main, the program runs fine, OpenOCD connects, and GDB can set and remove breakpoints as expected. 

However, I dislike the addition of the wait loop in main as it creates unnecessary complexity between debugging the program and running it normally, so I'd like to get rid of it. So, I've tried to start Spike with the `-H` flag and I've removed the while loop in main.

Unfortunately, I've run into some issues using GDB with this new setup: I can't seem to set breakpoints!
For completeness, I have attached the [OpenOCD log](https://github.com/riscv-software-src/riscv-isa-sim/files/10048128/openocd_log.txt) and [config file](https://github.com/riscv-software-src/riscv-isa-sim/files/10048109/spike.txt) (renamed because GitHub doesn't recognize .cfg)

These are the Spike commands I have been passing:
`$RISCV/bin/spike --rbb-port=5000 $RISCV/riscv64-unknown-elf/bin/pk $RISCV/hello`
`$RISCV/bin/spike --rbb-port=5000 -H $RISCV/riscv64-unknown-elf/bin/pk $RISCV/hello`

OpenOCD command:
`openocd -f spike.cfg`

GDB command and output:
`$RISCV/bin/riscv64-unknown-elf-gdb $RISCV/hello`
```
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x0000000000001000 in ?? ()
(gdb) b main
Breakpoint 1 at 0x10160: file hello.c, line 4.
(gdb) c
Continuing.
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x10160

Command aborted.
(gdb)
```

The program `hello` is as follows:
```c
#include <stdio.h>

int main() {
    printf("Hello World!\n");
}
```
Compiled with:
`$RISCV/bin/riscv64-unknown-elf-gcc -g hello.c -o hello`

I've noticed that I am able to run `monitor halt` and `monitor resume` perfectly fine in GDB, but even setting breakpoints with `monitor bp` doesn't seem to work, with the same error message. What am I doing wrong? Is there something I'm missing?

[This comment from issue #79](https://github.com/riscv-software-src/riscv-isa-sim/issues/79#issuecomment-267673917) seems to discourage using `-H` in favor of a while loop in main; is there a reason why? 

Thank you,
Eric

timsifive, 2022-11-21:
When you start spike halted, pk doesn't get a chance to run at all. pk needs to set up page tables before you can work with a program at 0x1000, which does not have memory behind it with the default spike settings.

scottj97, 2022-12-05:
Closing because the issue is apparently explained, and OP has not responded.

closed 2022-12-05 by scottj97