riscv-software-src/riscv-isa-sim #1163: Memory error on Hello World program

AshvinVaidyanathan, 2022-12-05:
I try to run the Hello World example using the  `/path/to/spike /path/to/pk hello`  command, but I get this error.

```
Access exception occurred while host was accessing memory on behalf of target (tohost = 0x1008000ea98):
Memory address 0x1008000ea98 is invalid
```

scottj97, 2022-12-05:
The problem is that your `hello` program uses addresses where Spike doesn't have any memory.

Two options: either change the way `hello` is compiled to use the default memory range supported by Spike (0x80000000-0xffffffff) or use the `-m` option to Spike to set up memory where you need it.

AshvinVaidyanathan, 2022-12-05:
> either change the way hello is compiled to use the default memory range supported by Spike (0x80000000-0xffffffff)

I know this is beyond the scope of this repo but could you elaborate a bit on how I could do that? I'm pretty new to this stuff and haven't used the compiler beyond the simple compiling and disassembling.

scottj97, 2022-12-05:
The easiest thing is to look at the disassembly of hello and see what addresses are used. Then use `-m` to put memories in those places in Spike.

To change `hello`'s memory map would (I guess?) require a custom linker script for the compiler toolchain.

AshvinVaidyanathan, 2022-12-07:
Hi @scottj97, I decided to reinstall spike and pk before I went ahead with your suggestion because I had installed the two a long time ago. Now it works perfectly fine. What could be the reason a clean install fixed it or rather why was the issue popping up in the first place. I know it's hard to know without actually seeing what was being done on my side, but I am just looking for any educated guesses. 

Thanks a lot for your help so far.

scottj97, 2022-12-07:
Sorry, no idea why that would have fixed it. Spike's default memory map hasn't changed in a long time, as far as I recall. I don't follow `pk` at all, so maybe something changed there.

closed 2022-12-07 by scottj97