riscv-software-src/riscv-isa-sim #1165: Non-contiguous hartids may cause Segmentation Fault

GuoShibo-cn, 2022-12-07:
I'm using spike with Non-contiguous hartids and find out that Non-contiguous hartids may cause Segmentation Fault. 
And by enlarging "hart_state" in debug_module.cc, bug fixed. 
I'm thinking the width of "hart_state" should not base on "nprocs" or proc_reset() should take "procs[i]" "i" as input instead of "hartid"
https://github.com/riscv-software-src/riscv-isa-sim/blob/adfaef00e5cd57bef0aa6a9909b4bff5b3863c40/riscv/debug_module.cc#L42-L45
https://github.com/riscv-software-src/riscv-isa-sim/blob/adfaef00e5cd57bef0aa6a9909b4bff5b3863c40/riscv/debug_module.cc#L935-L940
https://github.com/riscv-software-src/riscv-isa-sim/blob/adfaef00e5cd57bef0aa6a9909b4bff5b3863c40/riscv/processor.cc#L615-L617
https://github.com/riscv-software-src/riscv-isa-sim/blob/adfaef00e5cd57bef0aa6a9909b4bff5b3863c40/riscv/sim.cc#L450-L453
#156

aswaterman, 2022-12-07:
Clearly #165 wasn't sufficiently followed up.  Can you make a PR?

GuoShibo-cn, 2022-12-08:
#1166 is made for this issue

GuoShibo-cn, 2023-07-05:
Fixed by commit #1166 ,So issue could be closed.Thanks!

closed 2023-07-05 by GuoShibo-cn