riscv-software-src/riscv-isa-sim #1168: Avoiding dependencies on config.h

jerryz123, 2022-12-10:
Currently many spike-installed headers depend on the autoconf-generated `config.h`, which prevents usage of spike-as-a-library when including those headers.

I would like to remove these dependencies so I can just include `sim.h` in an external project and link with the spike library.
What is the canonical way to do this?

scottj97, 2022-12-11:
Linking Spike with outside code was enabled by a recent addition to this repo but is not well specified or well supported.

I link Spike with python but I use only a very small & simple C interface (that I added to my local fork of Spike) so I don't need any header files.

Spike compilation uses several configure-time choices such as `--enable-commitlog` and I assume this is why `config.h` is included. But perhaps the header files themselves don't need to care?

In the case of `sim.h` the interface changes if Boost libraries are available (`HAVE_BOOST_ASIO`). This has been a source of [issues](https://github.com/riscv-software-src/riscv-isa-sim/issues/834#issuecomment-944425960) in the past.

What if we moved all such `#ifdef` statements into the `*.cc` files and removed the include of `config.h` from the headers. Would that help you?

jerryz123, 2022-12-11:
> What if we moved all such #ifdef statements into the *.cc files and removed the include of config.h from the headers. Would that help you?

This would be great, and I'd be happy to submit some PRs towards this.

aswaterman, 2022-12-11:
If we can nuke all header dependencies on config.h, that would be the best outcome. If any headers truly must depend on config.h, they shouldn’t be installed. Thanks for helping improve this.

PR #1184 (2022-12-16): Support using sim.h/processor.h as public interfaces
This is the final set of patches towards enabling usage of sim.h and processor.h as public interfaces. Resolves #1168.

The primary danger with this change is dropping `config.h` off the include list in a file which references one of those variables, as the bug might only appear when a specific feature is enabled or disabled. To avoid this, I first add `#include "config.h"` directly to any source file which references any variables in `config.h`

The rest of these patches fix the installed headers list, change to relative include paths, and  [... cut]
closed 2022-12-16 by aswaterman