riscv-software-src/riscv-isa-sim #1172: Alignment  when 'C'  in misa CSR is cleared by software

liweiwei90, 2022-12-13:
Whether jump to 0x10082 triggers misaligned exception when  'C' in misa CSR is disabled by software? 
The answer is yes before Zc* is supported. 
However,  Zca is implied by 'C', but Zca can work alone without 'C' enabled.  So this becomes a new problem:
Whether Zca is also disabled when 'C' is disabled?

aswaterman, 2022-12-13:
The spec says the `misa` write is suppressed if the next instruction would not be IALIGN-bit aligned, so we have a few options here:

- Declare that if the implementation supports Zca, then it is unconditionally enabled.  The effect is that IALIGN is always 16 if Zca is supported, regardless of the setting of `misa.C`.  Therefore, clearing `misa.C` will always succeed, regardless of the PC.
- Declare that if the implementation supports C, then disabling C also disables Zca.  In this case, writes to `misa.C` should continue to be suppressed if the PC is not IALIGN-bit aligned.
- Change `misa.C` to be read-only, thereby avoiding this issue.

My preference is the middle option: if the implementation supports C, then clearing `misa.C` should disable Zca.  I think this follows the principle of least surprise.  But the other options are viable.

(Of course, if the implementation supports Zca but does not support C, then this issue is moot: the implementation always has IALIGN=16, because we have no means to disable Zca.)

aswaterman, 2022-12-14:
If Zcmp/Zcmt is implemented, `misa.C` must be hardwired to 0, so that is irrelevant.

Zcb requires Zca, so if we choose a strategy where clearing `misa.C` disables Zca, then clearing `misa.C` must also disable Zcb.

aswaterman, 2022-12-14:
Zcmp/Zcmt/Zcb require Zca, so if we choose a strategy where clearing `misa.C` disables Zca, then clearing `misa.C` must also disable Zcmp/Zcmt/Zcb.

closed 2023-02-07 by aswaterman

fanghuaqi, 2023-10-07:
Hi @liweiwei90 @aswaterman ，if this isa string is `rv32ima_zca_zcb_zcmp_zcmt/rv32imaf_zca_zcb_zcf_zcmp_zcmt/rv32imafd_zca_zcb_zcf_zcmp_zcmt/rv32imafd_zca_zcb_zcf_zcd` , will the `C` bit be set in misa csr? What is current spike implementation do?

fanghuaqi, 2023-10-11:
Hi @tariqkurd-repo , do you have any idea on this question https://github.com/riscv-software-src/riscv-isa-sim/issues/1172#issuecomment-1751667541 , which will affect Zc and C reflection on the CSR misa.C bit.

PR #1245 (2023-02-07): Make misa.C writable again, even in presence of Zc* extensions
Builds on and supersedes #1176. See extensive discussion on that PR.  Fixes #1172.

`misa.C` is now writable whenever `C` is included in the ISA string.  When `misa.C` is writable, all Zc* extensions are also under its direct control.  (There is still no mechanism to individually disable Zc* extensions, nor is there a mechanism to disable Zc* extensions when `misa.C` is read-only).

cc @liweiwei90, who authored the first version of this