riscv-software-src/riscv-isa-sim #1179: Spike error when following "Debugging With Gdb" section of README

andiluk, 2022-12-14:
Hi!

I tried to follow the "Debugging With Gdb" section of the README but Spike fails with:

```
$ spike --rbb-port=9824 -m0x10000000:0x20000 rot13-64
Listening for remote bitbang connection on port 9824.
Access exception occurred while loading payload rot13-64:
Memory address 0x10010000 is invalid
```

I encounter this error when using Spike built from the current master branch (SHA d00c01d...). This error does NOT occur when I use the build from the 1.1.0 release; there it works as expected!

When I debug Spike (`$ gdb --args spike --rbb-port=9824 -m0x10000000:0x20000 rot13-64`) I get following output right before the error occurs:

```
(gdb) info stack
#0  mmu_t::store_slow_path_intrapage (this=0x555555b6e5b0, addr=268500992, len=8, bytes=0x7fffffffcb80 "\267\a\001\020\203\247\307\t", xlate_flags=0, actually_store=<optimized out>) at ../riscv/mmu.cc:246
#1  0x00005555557c9cc0 in mmu_t::store<unsigned long> (xlate_flags=0, val=<optimized out>, addr=<optimized out>, this=<optimized out>) at ../fesvr/byteorder.h:61
#2  sim_t::write_chunk (this=<optimized out>, taddr=<optimized out>, len=<optimized out>, src=<optimized out>) at ../riscv/sim.cc:445
#3  0x0000555555a64de3 in memif_t::write (this=0x7fffffffcdc0, addr=<optimized out>, len=160, bytes=<optimized out>) at ../fesvr/memif.cc:82
#4  0x0000555555a6d44c in load_elf[abi:cxx11](char const*, memif_t*, unsigned long*, unsigned int) (fn=<optimized out>, memif=memif@entry=0x7fffffffcdc0, entry=entry@entry=0x7fffffffd570,
    required_xlen=<optimized out>) at ../fesvr/elfloader.cc:107
#5  0x0000555555a60ad9 in htif_t::load_payload (this=0x7fffffffd550, payload="rot13-64", entry=0x7fffffffd570) at /usr/include/c++/9/bits/basic_string.h:2304
#6  0x0000555555a61dfe in htif_t::load_program (this=0x7fffffffd550) at ../fesvr/htif.cc:147
#7  0x0000555555a6023c in htif_t::start (this=0x7fffffffd550) at ../fesvr/htif.cc:87
#8  0x0000555555a61051 in htif_t::run (this=0x7fffffffd550) at ../fesvr/htif.cc:227
#9  0x000055555577856b in main (argc=<optimized out>, argv=<optimized out>) at ../spike_main/spike.cc:554
(gdb) n

Catchpoint 1 (exception thrown), 0x00007ffff7e5b672 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) n
Single stepping until exit from function __cxa_throw,
which has no line number information.

Catchpoint 2 (exception caught), 0x00007ffff7e5a3e3 in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) info stack
#0  0x00007ffff7e5a3e3 in __cxa_begin_catch () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x0000555555774625 in htif_t::load_payload (this=0x7fffffffd550, payload="rot13-64", entry=0x7fffffffd570) at /usr/include/c++/9/ext/new_allocator.h:89
#2  0x0000555555a61dfe in htif_t::load_program (this=0x7fffffffd550) at ../fesvr/htif.cc:147
#3  0x0000555555a6023c in htif_t::start (this=0x7fffffffd550) at ../fesvr/htif.cc:87
#4  0x0000555555a61051 in htif_t::run (this=0x7fffffffd550) at ../fesvr/htif.cc:227
#5  0x000055555577856b in main (argc=<optimized out>, argv=<optimized out>) at ../spike_main/spike.cc:554
(gdb) n
Single stepping until exit from function __cxa_begin_catch,
which has no line number information.
htif_t::load_payload (this=0x7fffffffd550, payload="rot13-64", entry=0x7fffffffd570) at ../fesvr/htif.cc:140
140         bad_address("loading payload " + payload, t.get_tval());
(gdb) n
Access exception occurred while loading payload rot13-64:
Memory address 0x10010000 is invalid
[Inferior 1 (process 29308) exited with code 0377]
```

timsifive, 2022-12-15:
#889 put a UART at 0x10000000, which conflicts with this example. I'll fix the README to use a different address.

andiluk, 2022-12-15:
Ah I see, got it working now, thanks!

closed 2022-12-27 by aswaterman

omerguzelelectronicguy, 2024-01-16:
Is it possible to change UART by building spike? @timsifive 

I need to use `0x10000000` as starting program address. I Searched and find `NS16550_BASE`, `DM_DMCONTROL_ACKHAVERESET`, `elt_map` as platform specific address. I think `NS16550_BASE` is the parameter related to UART. Can I change it to another address and use that region to run a program?

PR #1196 (2022-12-27): Change address used in gdb debug example.
PR #889 put a UART at 0x10000000, which conflicts with the old address. Fixes #1179.
commit b282cafd (2022-12-27): Change address used in gdb debug example.

PR #889 put a UART at 0x10000000, which conflicts with the old address.
Fixes #1179.