riscv-software-src/riscv-isa-sim #1188: Issue facing with LD instruction

shraddha10101, 2022-12-20:
[LD.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/10265646/LD.txt)

Hi, we are trying to run LD instruction in spike. When we run this program we expect to see t5 = 0x36 and t6 = 0x20000 finally. But when we add the 54th and 55th testcase, we are not able to see the same(when I say testcase, I mean both the la and ld instruction).
What could be the possible explanation? Any help is appreciated. 

Thanks!

scottj97, 2022-12-20:
Have you run Spike with `-l --log-commits` and looked to see what's happening?

aswaterman, 2022-12-21:
I believe the problem is your choice of registers.  In the standard ABI, `x3` is the global pointer, and linker relaxations from `la` might use it.  So, overwriting `x3` without special precautions is disallowed in the standard ABI.

Going forward, I advise debugging your code, including using the method Scott suggested, before raising a problem with Spike.

shraddha10101, 2022-12-22:
Thank you for your valuable input. We will look into it by following the method that Scott suggested. Thanks.

closed 2022-12-22 by shraddha10101