riscv-software-src/riscv-isa-sim #1215: To trigger interrupt in interaction mode

sbgssol, 2023-01-06:
Hello,

Can someone give me a guideline or references to:
(A) trigger or request an interrupt while executing instruction in the interaction mode? or,
(B) give Spike where (PC addresses) I want to place some interrupts before executing instructions, then whenever Spike reaches the preset PC, it automatically raises an interrupt.

My intention for `(A)`:
```
$ ./spike -d out.elf
: r 1
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
: r 1
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
: r 1
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
: raise interrupt
core   0: 0x000000000000100C # Interrupt here!
```

About `(B)`
```
addresses = [0x1004, 0x100c]
./spike --interrupt=addresses -l out.elf
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 # Interrupt here!
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100C # Interrupt here!
```

Thanks!

aswaterman, 2023-01-06:
These features don’t exist in Spike.

sbgssol, 2023-01-06:
Thanks for your info, 
I intended to trigger an interrupt externally (by not using Zicsr instruction to modify the MIP and MIE).
So I understand that there is no way to trigger an interrupt during the interaction mode or using available spike commands.

WhiteBin, 2024-05-16:
Has it been successfully implemented?

sbgssol, 2024-05-16:
> Has it been successfully implemented?

Hi @WhiteBin , I have not touched this topic for a year, but the approach (A) worked.
As far as I remember, by adding some custom commands, which directly write expected values to the corresponding registers (need to read the spec about priority and its mechanism). It could be used it in the spike command file, which was great for controlling the interrupt/break based on the PC address, unfortunately, I cannot share more details. Hope it help!

WhiteBin, 2024-05-16:
您好，邮件已收到，谢谢

WhiteBin, 2024-05-17:
> > Has it been successfully implemented?
> 
> Hi @WhiteBin , I have not touched this topic for a year, but the approach (A) worked. As far as I remember, by adding some custom commands, which directly write expected values to the corresponding registers (need to read the spec about priority and its mechanism). It could be used it in the spike command file, which was great for controlling the interrupt/break based on the PC address, unfortunately, I cannot share more details. Hope it helps!

Thank you very much！ I am trying to follow this train of thought