riscv-software-src/riscv-isa-sim #1221: [question] which debug spec version is supported by spike?

aap-sc, 2023-01-11:
Hi!

Currently, readme says:

```
+Spike supports the following RISC-V ISA features:
...
+  - Debug v0.14
```

However, I can't find 0.14 anywhere. I've seen the following documents:

- https://riscv.org/wp-content/uploads/2019/03/riscv-debug-release.pdf (which is 0.13.2)
- https://raw.githubusercontent.com/riscv/riscv-debug-spec/master/riscv-debug-stable.pdf (which is deemed to be 1.0)

Could someone clarify the situation? Maybe @aswaterman (since git blame shows you as the author) or @timsifive ?

timsifive, 2023-01-11:
It should say 1.0 stable. I wouldn't swear that spike does everything according to the spec, but that's the goal.

aswaterman, 2023-01-11:
If you think we should bump the version number in the README, can you PR it?

timsifive, 2023-01-11:
Yes, already working on it.

aap-sc, 2023-01-12:
@timsifive , thanks for a prompt response!

closed 2023-01-12 by aap-sc