riscv-software-src/riscv-isa-sim #1226: LR.D/SC.W issue, example lr.d begin address is 0x0, and then sc.w begin address is 0x4, should the sc success ???

zack-zw, 2023-01-13:
from spec, it seems this scene should success.

[image]

but from spike mmu.h, sc is fail due to return value is judge from expression (load_reservation_address == paddr)

from mmu.h line 182
inline bool check_load_reservation(reg_t vaddr, size_t size)
  {
    if (vaddr & (size-1)) {
      // Raise either access fault or misaligned exception
      store_slow_path(vaddr, size, nullptr, 0, false, true);
    }

    reg_t paddr = translate(vaddr, 1, STORE, 0);
    if (sim->reservable(paddr))
      return load_reservation_address == paddr;
    else
      throw trap_store_access_fault((proc) ? proc->state.v : false, vaddr, 0, 0);
  }

aswaterman, 2023-01-13:
See the "eventual success" section later in the chapter.  https://github.com/riscv/riscv-isa-manual/blob/2c9b100696ea44fa0949fd306bb42769a2c84d73/src/a.tex#L305-L306

The SC is only guaranteed to eventually succeed if it has the same address and size.  It might succeed on some attempts on some implementations, but it might never succeed on others.  Spike's behavior here is legal.

zack-zw, 2023-01-13:
> See the "eventual success" section later in the chapter. https://github.com/riscv/riscv-isa-manual/blob/2c9b100696ea44fa0949fd306bb42769a2c84d73/src/a.tex#L305-L306
> 
> The SC is only guaranteed to eventually succeed if it has the same address and size as the LR. It might succeed on some attempts on some implementations, but it might never succeed on others. Spike falls into the latter category; its behavior is legal.

Got it. Thanks a lot.

aswaterman, 2023-01-13:
You’re welcome!