riscv-software-src/riscv-isa-sim #1227: Using spike with RV32 isa and RoCC extension (--extension)

JEvSilv, 2023-01-13:
Hello,

I'm performing spike with flags `--isa=RV32IMAFDC --exentension=<ext>` and it returns: "An illegal instruction was executed!". I found that in the function `rocc_t::get_instructions()` RV32 functions (insns_desc_t) are assigned to `&::illegal_instruction`. So I replaced `&::illegal_instruction` with custom functions (c0-c3) and It worked. There is some reason to restrict extensions just for RV64?

Thank you!

aswaterman, 2023-01-13:
When RoCC was originally created, it was only used with RV64, so there’s a historical element.

Spike’s RoCC templates were only designed for RV64 and don’t enforce that data written back to the x-registers is in canonical 32-bit sign-extended form for RV32. There might be other issues along those lines. But I don’t think they’re fundamentally impossible to fix.

JEvSilv, 2023-01-13:
Thank you for your response!

closed 2023-01-13 by JEvSilv