riscv-software-src/riscv-isa-sim #1231: Multiple trigger match priority

scottj97, 2023-01-16:
In the debug spec 5.3 "Priority" it says about multiple triggers firing simultaneously:

> If one of these triggers has the “enter Debug Mode” action (1) and another trigger has the “raise a breakpoint exception” action (0), the preferred behavior is to have both actions take place. [...] If this is not implemented, then the hart must enter Debug Mode and ignore the breakpoint exception.

The current code in [triggers::module_t::detect_memory_access_match](https://github.com/riscv-software-src/riscv-isa-sim/blob/e4d2458e7eac7b7f098f225be69c9af224c930ed/riscv/triggers.cc#L477-L496) and [triggers_t::module_t::detect_trap_match](https://github.com/riscv-software-src/riscv-isa-sim/blob/e4d2458e7eac7b7f098f225be69c9af224c930ed/riscv/triggers.cc#L504-L508) does not conform to this. It will return the match from the lowest-index trigger (or chain of triggers) that matches.

(The new code for icount triggers in #1220 has the same problem.)

Does this also have a problem if one matching trigger has timing=before and another has timing=after? Seems like the timing=before triggers must take priority.

cc @YenHaoChen

scottj97, 2023-01-16:
@YenHaoChen has added commit fb8a8d5cbc6665e9dd89961dab2cf24c33d1ed80 to #1220 which addresses the dmode issue.

Is timing=before/after also an issue?

YenHaoChen, 2023-01-17:
The debug spec defines the priority of timing=after on previous instruction has a higher priority than the timing=before on current instruction in Table 5.2. Thus, if the previous instruction matches timing=after, the current instruction does not need to check triggers with timing=before. 

[image]

However, the table does not constrain the behavior of an instruction matching multiple triggers with both timing=before and timing=after. Thus, https://github.com/riscv-software-src/riscv-isa-sim/commit/fb8a8d5cbc6665e9dd89961dab2cf24c33d1ed80 does not consider the timing when choosing a trigger action. 

@timsifive, would you help to clarify if this behavior is desired?

scottj97, 2023-01-17:
My concern is when trigger 1 matches with timing=after, and trigger 2 matches with timing=before, on the same instruction.

Trigger 1 by itself would complete the instruction and then take the breakpoint trap afterward. It will be the highest-priority exception in Table 5.2, but not until the next instruction.

Trigger 2 by itself would not complete the instruction, taking the breakpoint trap instead.

When both match, it seems to me that the instruction should not complete, since trigger 2 should cause an immediate trap. But if I'm reading the code correctly, that's now how it works today in Spike (either current master of a7ac8d85a0f3dcabf181c1858e0e7d02a6cd71e1 or fb8a8d5cbc6665e9dd89961dab2cf24c33d1ed80 in #1220). It will complete the instruction, ignoring trigger 2, and then take the trap for trigger 1.

scottj97, 2023-01-26:
@timsifive we would appreciate your opinion on priority when triggers with timing=before and timing=after both match on the same instruction.

timsifive, 2023-01-26:
> My concern is when trigger 1 matches with timing=after, and trigger 2 matches with timing=before, on the same instruction.
> When both match, it seems to me that the instruction should not complete, since trigger 2 should cause an immediate trap.

I agree. If the before trigger matches, then the trigger action is taken before the instruction retires. Since the instruction doesn't retire, the after trigger cannot match.

Note that (at least for mcontrol6) I'm proposing a spec change that lets the target pick whatever timing it wants. (https://github.com/riscv/riscv-debug-spec/pull/795) It's unclear if people will agree with me.