riscv-software-src/riscv-isa-sim #1236: Cannot modify memory by external libraries/devices

Z-KN, 2023-01-23:
(1) Following the example of #315, I wonder how to modify values in memory by external libraries? External devices sometimes can produces results that is required by the RISCV core. But through given functions of `load` and `store`, I do not know how to do that.

(2) Another problem is that, it seems any `lw` will load 0 from any address, even though that address's value was stored using `sw`  by CPU before.

```
# Do a load and a store of address 0x10000000 to exercise the MMIO plugin.
li t0, 0x10000000
li t1, 0x44332211
sw t1, 0(t0)
lw t2, 0(t0)
```
When I `printf("BYTES=%x\n", *bytes);` in `test_mmio_plugin_load`, the results are always `0`, which I think `11` is expected. Is it normal?

scottj97, 2023-01-23:
IIUC, the MMIO plugin interface is for you to implement a device that is not just a simple RAM. The `bytes` parameter is a pointer, and your custom `load` function is expected to modify the bytes pointed to. `*bytes` will likely contain garbage upon function entry.

Z-KN, 2023-01-23:
> IIUC, the MMIO plugin interface is for you to implement a device that is not just a simple RAM. The `bytes` parameter is a pointer, and your custom `load` function is expected to modify the bytes pointed to. `*bytes` will likely contain garbage upon function entry.

I tried in the following way:
I modify `load` as follows:
```
bool test_mmio_plugin_load(void* self, reg_t addr, size_t len, uint8_t* bytes)
{
    printf("LOAD -- SELF=%p ADDR=0x%lx LEN=%lu BYTES=%x\n", self, addr, len, *bytes);
    *bytes=0x44;
    printf("LOAD -- SELF=%p ADDR=0x%lx LEN=%lu BYTES=%x\n", self, addr, len, *bytes);
    return true;
}
```

And I test like the following:
```
# Do a load and a store of address 0x10000000 to exercise the MMIO plugin.
li t0, 0x10000000
lw t1, 0(t0)
lw t2, 0(t0)
```
The results show:
```
ALLOC -- ARGS=argument
LOAD -- SELF=0x123456789abcdef ADDR=0x0 LEN=4 BYTES=0
LOAD -- SELF=0x123456789abcdef ADDR=0x0 LEN=4 BYTES=44
LOAD -- SELF=0x123456789abcdef ADDR=0x0 LEN=4 BYTES=0
LOAD -- SELF=0x123456789abcdef ADDR=0x0 LEN=4 BYTES=44
DEALLOC -- SELF=0x123456789abcdef
```
After 'modifying', next time when loading the value from the same address (0x10000000), the register still gets `0`. So I think this means using pointers to direct modify values through `load` function is not feasible. Or what am I missing?

scottj97, 2023-01-23:
What you're missing is that Spike is not supposed to remember your device's state -- that's what your device is supposed to do. If you want a load from the CPU to see the data most recently stored, then it's up to your device to take the data given to it by the `store()` function, save it, and return that on `load()`.

You should see the 0x44 being written to the destination of the `lw` (`t1` in your first load). However since you're only writing one byte within `load()`, the other three bytes will still be garbage. You should modify not only `*bytes` but also `*(bytes + 1)` etc. up to `len` bytes.

Z-KN, 2023-01-23:
> 

Thank you for the suggestion!

IIUC, I tried
```
bool test_mmio_plugin_store(void* self, reg_t addr, size_t len, const uint8_t* bytes){
//statement 1
    uint8_t *addr_mod=addr+0x4; // assume address of return reg
    *addr_mod=0X55; // assume some computation results
//statement2
    uint8_t *addr_mod=bytes+0x4;
    *addr_mod=0X55;
}
```
but either of two statements raises `Segmentation fault`. So as you pointed out, if I want to modify values in device memory space, in which function should I do that modification?

scottj97, 2023-01-23:
Why on earth are you trying to write to `addr+0x4` from within `store()`? That `addr` is the target address, within the simulated RISC-V world, and you're using it as a host address. Also, there is no "return reg" for a store operation; I have no idea what that's supposed to mean.

Why on earth are you trying to write to `bytes+0x4` from within `store()`? That `bytes` points to the data to be stored, and only the first `len` bytes are valid.

If you need memory in which to save state, it's up to your MMIO device to allocate that memory, probably by calling `malloc()` from within its `alloc` function.

Or for a cleaner, safer implementation, use the C++ API and simply declare some attributes within your class instance.

Untested example:
```
struct my_mmio_device {
    bool load(reg_t addr, size_t len, uint8_t* bytes) {
        if (len != 4) return false;  // only 4 byte loads supported
        memcpy(bytes, &mydata, 4);
        return true;
    }
    bool store(reg_t addr, size_t len, const uint8_t* bytes) {
        if (len != 4) return false;  // only 4 byte stores supported
        memcpy(&mydata, bytes, 4);
        return true;
    }
    uint32_t mydata = 0x11223344;
};
```

Z-KN, 2023-01-23:
> 

Thank you very much! Previously I tried to write values to `byte` directly but was refused, while `memcpy` works. Some of the address management scheme is different from that of an SoC, which confuses me a little. I am grateful to your detailed and patient explanations.

closed 2023-01-25 by Z-KN