riscv-software-src/riscv-isa-sim #1238: "User load/store segfault @ <addr>" arises when executing C programs

Z-KN, 2023-01-25:
I modified the source code of Spike and managed to add a device, inheriting `abstract_device_t`, to Spike, with a certain base address, say, `0x400000000`. Then I built Spike again.

When I write assembly file `test.S` as
```
...
li t0, 0x400000000
li t2, 0x44332211
sw t2, 0(t0)
lw t1, 4(t0)
sw t1, 0(t0)
...
```
and compile with `riscv64-unknown-elf-gcc -g -nostdlib -static -Wl,-Ttext-segment,0x80000000 test.S -o test-asm` and simulate with `spike test-asm`, the simulator shows correct results (triggering `load` and `store` functions implemented by me in before).

However, when I try to move the assembly code to C program (`test.c`) as follows:
```
#include "stdio.h"
int main(){
	asm volatile(
"li t0, 0x400000000;"
"li t2, 0x44332211;"
"sw t2, 0(t0);"
"lw t1, 0(t0);"
"sw t1, 4(t0);"
    );
}
```
or
```
#include "stdio.h"
int main(){
volatile unsigned int *ptr;
ptr=(unsigned int *)0x400000000;
*ptr=0x44332211;
}
```
and compile with `riscv64-unknown-elf-gcc -static test.c -o test-c` and simulate with `spike pk test-c`. the results will generate errors as
```
...
pc 00000000000101bc va/inst 0000000400000000 sr 8000000200006020
User store segfault @ 0x0000000400000000
```

So the case is that the assembly code works while the C program does not. I think perhaps users cannot directly access to the device memory using C code? Then I wonder how to make C work? Do I need to add some system calls like `mmap()` in test C file or do additional modifications to Spike source files?

scottj97, 2023-01-25:
This is not a Spike problem, I am sure. Spike is faithfully executing the program you gave it, but for some reason 0x40000000 is not accessible to the program.

I don't know anything about `pk` or the C runtime, but if you run Spike with `-l --log-commits` you should be able to see in detail what's happening before that PC value of `101bc`. Perhaps PMP is being programmed to disallow access to that address?

Z-KN, 2023-01-27:
> This is not a Spike problem, I am sure. Spike is faithfully executing the program you gave it, but for some reason 0x40000000 is not accessible to the program.
>
> I don't know anything about `pk` or [...]

Yes. I figured out that it is `pk` that limits the RISCV running in the user mode, under which user can only access main memory address, not including MMIO devices. Thank you!

closed 2023-01-27 by Z-KN