riscv-software-src/riscv-isa-sim #1239: Svadu support approach

adurbin-rivos, 2023-01-27:
I'm working on adding in the Svadu support, and I wanted to get an opinion on best approach. Currently, there's an --mmu-dirty flag which effectively acts like menvcfg.HADE=1 and henvcfg.HADE=1. So the questions I have are:

1. Should we maintain the --mmu-dirty flag in its current form as a standalone behavior.
2. When --mmu-dirty flag is set force {menvcfg,henvcfg}.HADE=1 in the backing register state and don't let them be writable. Set bits are exposed.
3. Same as 2 but mask of HADE being set.
4. Always enable the optionality of the HADE bits no matter what and rely on the runtime of the state to decide behavior.
5. Same as 4 but supply a flag which enables HADE behavior to be exposed.

If I missed any options feel free to suggest as I'm open to whatever. Thank you.

aswaterman, 2023-01-27:
My recommendation is to ditch `--mmu-dirty` altogether.  (It has only been a part of the codebase for a few weeks, so it won't be particularly disruptive to nuke it.)

Then, if `Svadu` is not part of the ISA string, the various `HADE` fields are hardwired to 0.  If it is part of the ISA string, then those fields become writable (but should reset to 0).

cc @jerryz123, who added the `--mmu-dirty` support recently.

jerryz123, 2023-01-27:
I have no attachment to that flag. Previously, mmu-dirty was compile-time configuration option, I only changed it to be a runtime flag.

adurbin-rivos, 2023-01-27:
Perfect. Thanks for the direction. I'll proceed accordingly. And keep this issue open while it's being worked on in case I have other comments/questions.

aswaterman, 2023-01-27:
Cool. I’ll review your PR when it’s ready.

adurbin-rivos, 2023-01-27:
I don't know how to link a PR and an issue on the web as I am GitHub stunted. Anyway, I threw this one together that still needs to be tested: https://github.com/riscv-software-src/riscv-isa-sim/pull/1240

aswaterman, 2023-01-28:
@adurbin-rivos For future reference: GitHub detects strings like "Resolves #1239" in PR descriptions.  Including that one will automatically close the mentioned issue when the PR is merged.

aswaterman, 2023-01-28:
Addressed by #1240

closed 2023-01-28 by aswaterman