riscv-software-src/riscv-isa-sim #1248: Spike jump different adress than I expected.

omerguzelelectronicguy, 2023-02-10:
The title may not explain my problem well. sorry for that.
I am using spike and RTL simulation to match and verify the RTL is working correctly. my verification is low-level for now. I don't need system features. So, I am just adding the assembly to the CRT file after running some phases. Those might not be a good way to make such verification(I am open to suggestions). The problem started after I rebuild the compiler and spike. Spike simulation started to jump to another part of the program that I don't know. Previously simulation was starting with the bootloader and then goes CRT instructions (As I know I didn't change the linker file).
This is possibly not a real problem but I miss some points. And I need to solve or understand well to make my flow better. Could you help me, please? 

compilation command:
```
/diskd/riscv-gnu-toolchain/release/bin/riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -fno-tree-loop-distribute-patterns -o I_3.s.riscv ../test_code/empty_main.c ../lib/syscalls.c ../lib/crt_with_asm.S -static -nostdlib -nostartfiles -lm -lgcc -T ../lib/test.ld
```

scottj97, 2023-02-10:
There's not enough information here to diagnose, but this is very unlikely to be a Spike problem, so this is not the best place to ask. The [sw-dev](https://groups.google.com/a/groups.riscv.org/g/sw-dev) group would get you better answers.

Try disassembling the output file to see where the code is going. Or create a map file by adding `-Wl,-Map=foo.map` and see where the linker is putting things. Run Spike with `-l --log-commits` to see the program flow.

closed 2023-02-10 by scottj97

omerguzelelectronicguy, 2023-02-13:
I actually know where it goes but I don't understand why it goes differently than before. I mean of course, it is in the linker but previously it wasn't starting from there. I just thought that it might be a common confusion that is caused by a flag. Seems I am wrong. That is ok. I can solve it by using some other ways.

Additionally, the problem is I am not familiar with and having enough knowledge of the compilation and system support etc. to run a simple assembly. spike requires a compiler (to make executable), either pk or some part of codes that makes the pk's job. I am a designer of hardware that should be verified with some golden model. Spike is one of the best options for this. Thanks to the developers. 
There are many small issues that are caused by the lack of that system knowledge of people who just want to verify their cores with the assembly code just they wrote it. This feature can prevent many dummy issues (like mine) so that you can deal with real developing issues.

scottj97, 2023-02-13:
> This feature can prevent many dummy issues (like mine) so that you can deal with real developing issues.

What feature? You aren't asking for any feature! Instead you opened an issue on Spike's issue tracker, intended for reporting bugs in this golden model, asking for help with your compiler toolchain. As I explained, there are much more appropriate places to seek such help.