riscv-software-src/riscv-isa-sim #1253: why update context when change plic threshold?

WadeGao, 2023-02-17:
I have some questions about the function `bool plic_t::context_write(plic_context_t *c, reg_t offset, uint32_t val)` in riscv/plic.cc

when `offset == CONTEXT_THRESHOLD` and  `val > max_prio`, we set `update` and call `context_update` later. bu nothing of the current context changed. 

why update the context is necessary when meeting this case? 

[image]

aswaterman, 2023-02-17:
Generally, we prefer references to specific lines of code, rather than screenshots, which destroy much of the information.

@avpatel FYI this seems like your block

avpatel, 2023-02-17:
@WadeGao This looks like a bug which was introduced unintentionally. The "update" should be set whenever the context threshold is written so "update = true" should be under "if ()" and not under "else".

WadeGao, 2023-02-17:
> Generally, we prefer references to specific lines of code, rather than screenshots, which destroy much of the information.
> 
> @avpatel FYI this seems like your block

Thank you, I am afraid of that there may be some change of this file during our discussion. I will pay more attention to the open source community rules next time.

closed 2023-02-18 by WadeGao