riscv-software-src/riscv-isa-sim #1255: where is source code location to load riscv executable file?

dshm, 2023-02-20:
where is source code location to load riscv executable file? For example, running `spike pk helloworld`, where is source code location to load riscv executable file helloworld? I search the source code for many times and only find the location to load pk at htif.cc htif_t::start(). htif_args for sim_t  constructor and targs for hitf_t class save the path of pk and helloworld, but I can't find the location to load riscv executable file helloworld. Is there some advice? thank you very much!

aswaterman, 2023-02-21:
It's part of `pk`.  https://github.com/riscv-software-src/riscv-pk

dshm, 2023-02-21:
> 

which file in pk? Could you please give some tips? I need to find it as soon as possible.@aswaterman

dshm, 2023-02-21:
> It's part of `pk`. https://github.com/riscv-software-src/riscv-pk

which file in pk? Could you please give some tips? I need to find it as soon as possible.@aswaterman

aswaterman, 2023-02-21:
You need to bear in mind that supporting this software is a volunteer effort. I’d ask that you not try to rush people along. Anyway, https://github.com/riscv-software-src/riscv-pk/blob/master/pk/elf.c