riscv-software-src/riscv-isa-sim #1258: multi core reservation set support

hanfeng0114, 2023-02-21:
Hi

It seems right now the lr/sc reservation set is only supported within one processor/multi threads scenario.
But for multi processors scenario, there is no reservation set implementaion.
If we want to emulate 2 cluster, 4 processors/cluster scenario, we can only set spike as 1 processor with 8 threads?
Is it expected?

Thanks

aswaterman, 2023-02-21:
Spike's implementation is correct for any number of harts.  How these harts are distributed amongst cores is not relevant to the correctness of the implementation.

The rationale is that Spike executes several instructions on one hart before switching to another hart.  So, it will make progress through any LR/SC sequence that meets the eventuality requirements: https://github.com/riscv/riscv-isa-manual/blob/b39b1bec376ee661bfb05eebe4e7ea781fe4d803/src/a.tex#L280

aswaterman, 2023-02-21:
(Put another way: if having 4 cores with 2 harts apiece affects LR/SC forward progress vs. having 8 cores with 1 hart apiece, then you might have a bug in your implementation.)

closed 2023-02-21 by hanfeng0114