riscv-software-src/riscv-isa-sim #1271: Question about riscv vector floating-point comparison instruction.

lico614257, 2023-03-06:
from riscv vector spec1.0, sNaN inputs will raise the invalid operation exception in vmfeq and vmfne instruction.

13.13. Vector Floating-Point Compare Instructions
These vector FP compare instructions compare two source operands and write the comparison result to a mask register. The
destination mask vector is always held in a single vector register, with a layout of elements as described in Section Mask
Register Layout. The destination mask vector register may be the same as the source vector mask register (v0). Compares
write mask registers, and so always operate under a tail-agnostic policy.
The compare instructions follow the semantics of the scalar floating-point compare instructions. vmfeq and vmfne raise the
invalid operation exception only on signaling NaN inputs. vmflt, vmfle, vmfgt, and vmfge raise the invalid operation
exception on both signaling and quiet NaN inputs.

but it not raise any invalid operation exception in my machine.the following is my test code:

u64 rd[2];
double r1[2] = {SNANL, SNANL};
double r2[2] = {SNANL, SNANL};
__asm__ volatile(
    "vsetivli a1, 2, e64, m1, ta, ma     \n"
    "vle64.v       v1, (%0)                   \n"
    "vle64.v       v2, (%1)                   \n"
    "vmfeq.vv      v4, v2, v1, v0.t      \n"
    "vse64.v       v4, (%2)                 \n"
    :
    : "r"(r1),"r"(r2), "r"(rd));

lico614257, 2023-03-07:
I have initialized the mask register to contain 1s.but it also don’t raise invalid operation exceptions. the following is my test code: 
---------------------------------
    u64 v0_mask[2] = {0xf, 0x0};
    u64 rd[2];
    double r1[2] = {SNANL, SNANL};
    double r2[2] = {SNANL, SNANL};
    __asm__ volatile(
        "vsetivli a1, 2, e64, m1, ta, ma     \n"
        "vle64.v       v0, (%0)                   \n"
        "vle64.v       v1, (%1)                   \n"
        "vle64.v       v2, (%2)                   \n"
        "vmfeq.vv      v4, v2, v1, v0.t       \n"
        "vse64.v       v4, (%3)                  \n"
        :
        : "r"(v0_mask),"r"(r1),"r"(r2), "r"(rd));

aswaterman, 2023-03-07:
There are additional problems with this code:
- `SNANL` is a long double, not a double.  The implicit conversion to double will convert the signaling NaN into a quiet NaN.
- The compiler is not required to initialize any of those arrays, because the inline asm does not depend on them.  (Depending on the address of an array is not the same thing as depending on the array.)  Making all of the arrays volatile would work around that problem.

In any case, Spike correctly raises exceptions on signaling NaN arguments to `vmfeq`.  But going forward, please only raise support tickets on Spike when Spike is to blame.  You need to first debug your code and make sure it's doing what you think it's doing.

closed 2023-03-07 by aswaterman