riscv-software-src/riscv-isa-sim #1278: Why spike does not work with both large static or dynamic arrays?

NazerkeT, 2023-03-08:
Hi there,

I was recently heavily using spike with pk and noticed that I cannot run application with large mallocs like 8MB or more with spike + pk any more. The same goes when I try to run application with static arrays of 1024x1024. Essentially, simulation fails with either load or store segfaults.

Is there any reason that is coming from spike side for it? If yes, how can I change the current spike limitations for it? In general, is there any hint or advice do you guys have for it? 

Thanks,
Best wishes

NazerkeT, 2023-03-08:
Thanks for your response, @aswaterman ! 

however, do you mean here gcc compiler or os in general? because, essentially what I am doing is just statically compiling my app into elf and running a simulation. 

if we are talking about gcc or toolchain versions, which versions do you recommend to use?

aswaterman, 2023-03-08:
Not the toolchain itself, but things like the C library (which manages the heap) and the OS or OS-like layer (which provides the memory for the heap).  `riscv-pk`, which is a simple runtime for a single process, handles large heaps fine.  For example:

```
$ cat test.c
#include <stdio.h>
#include <stdlib.h>

int main()
{
  unsigned n = 268435456;
  volatile int* array = (int*)malloc(n * sizeof(int));

  for (unsigned i = 0; i < n; i++) {
    array[i] = i + 1;
  }

  unsigned long long sum = 0;
  for (unsigned i = 0; i < n; i++) {
    sum += array[i];
  }

  printf("sum of 1 to %u is %llu\n", n, sum);

  return 0;
}
$ riscv64-unknown-elf-gcc -O2 test.c
$ spike pk a.out
bbl loader
sum of 1 to 268435456 is 36028797153181696
$
```

NazerkeT, 2023-03-09:
hmm, interesting, the same app runs just fine for me, will try to debug more for exact reasons of why my app is not running. Thanks!

Another side question is, is it possible to run spike in parallel like we can do for make? For example, in teh same manner when we do `make -j20`, can we do `spike -j20` for log operations specifically? Or is spike highly serial for logging?

Thanks,
Best wishes,

closed 2023-03-09 by aswaterman