riscv-software-src/riscv-isa-sim #1282: when 0<vstart<vl , executing vfadd.vv instruction,  trap_illegal_instruction occurs

Fushiyu1, 2023-03-15:
The assembler is as follows:

```
    li a0 , 0x3;
    vsetvli a3 , a0 , e32 , m1 , ta , ma;
    csrr t2 , vl;
    li a0 , 0x1;
    csrw vstart , a0;
    vfadd.vv v6 , v3 , v4;
```
The log is as follow:
```
core   0: 0x000000000001063c (0x00300513) li      a0, 3
core   0: 0x0000000000010640 (0x0d0576d7) vsetvli a3, a0, e32, m1, ta, ma
core   0: 0x0000000000010644 (0xc20023f3) csrr    t2, vl
core   0: 0x0000000000010648 (0x00100513) li      a0, 1
core   0: 0x000000000001064c (0x00851073) csrw    vstart, a0
core   0: **0x0000000000010650** (0x02321357) vfadd.vv v6, v3, v4
core   0: exception trap_illegal_instruction, epc **0x0000000000010650**
core   0:           tval 0x0000000002321357
core   0: >>>>  trap_vector
```

In the require_vector function, there are two variables to limit whether vstart is required to be 0, but we did not understand the meaning of vstart_alu.
```
    if (alu && !P.VU.vstart_alu) \
      require(P.VU.vstart->read() == 0); \
```
And at riscv-v-spec , vfadd instruction doesn't require that vstart is 0,  so why did trap_illegal_instruction occur?

Liuziying04, 2023-03-15:
I have the same problem, Why does spike behave differently from the spec manual?

aswaterman, 2023-03-15:
This behavior is entirely consistent with the spec:

> Implementations are permitted to raise illegal instruction exceptions when attempting to execute a vector instruction with a value of vstart that the implementation can never produce when executing that same instruction with the same vtype setting.

> NOTE: For example, some implementations will never take interrupts during execution of a vector arithmetic instruction, instead waiting until the instruction completes to take the interrupt. Such implementations are permitted to raise an illegal instruction exception when attempting to execute a vector arithmetic instruction when vstart is nonzero.

closed 2023-03-15 by aswaterman

aswaterman, 2023-03-15:
For completeness, I will note that other behaviors are legal.  Executing the instruction in the obvious way, starting from vstart, is also a valid implementation decision.

We chose to adopt the most conservative implementation in Spike for two reasons: it's simple, and it discourages software developers from misusing `vstart`.