riscv-software-src/riscv-isa-sim #1283: Why logging stops after 513MB?

NazerkeT, 2023-03-15:
Hi there,

Sorry for asking so many questions recently!

However, while I was testing my application I noticed that spike's logging stops at 513MBs exactly for any application log that is presumably larger than that with this command: `spike -l --log=app.log --isa=rv64gc pk app`. So far I looked at the `limits` under `proc` while the program was running and it seems that there is now onvious limitation over there. Do you have any hints on how I can solve or debug this problem?

Thanks,
Best wishes,

aswaterman, 2023-03-15:
That's definitely not intentional.  I'm unable to replicate it on my machine; Spike happily spits out arbitrarily long logs for me.

NazerkeT, 2023-03-20:
> That's definitely not intentional. I'm unable to replicate it on my machine; Spike happily spits out arbitrarily long logs for me.

Got it, thanks! Unfortunately, I could not figure out that problem yet and instead using some workarounds. Looks like it is the issue coming from my device and so far will close this issue, until I have better clarification questions :)

Best wishes,

closed 2023-03-20 by NazerkeT