riscv-software-src/riscv-isa-sim #1287: Why different behavior occurs on the same testcase between spike and qemu?

WadeGao, 2023-03-17:
I tested `rv64ui-v-add` in the `riscv-tests` offical testcases repo. The elf file `rv64ui-v-add` was built on the latest commit(c825b1f1678be926b226c74a9ed997a4ea99db98).

But the instruction flows seemed very different between qemu and spike. Spike finished the execution successfully
[image]
but qemu was trapped in an endless illegal instruction address range (like 0xffffffffffe022bc). Although spike also met the same illegal instruction address range, but it finished successfully with code 0

Belows is my record of running testcase with qemu: 
```bash
$ qemu-system-riscv64 -M virt -m 16M -kernel rv64ui-v-add -bios none -serial stdio -display none -s -S
$
$ riscv64-unknown-elf-gdb rv64ui-v-add
Reading symbols from rv64ui-v-add...
(No debugging symbols found in rv64ui-v-add)
(gdb) target remote localhost:1234
Remote debugging using localhost:1234
0x0000000000001000 in ?? ()
(gdb) b *0x80000140
Breakpoint 1 at 0x80000140
(gdb) continue
Continuing.

Breakpoint 1, 0x0000000080000140 in pop_tf ()
(gdb) ni
0x0000000000002988 in ?? ()
(gdb) ni
0xffffffffffe00144 in ?? ()
(gdb) ni
0xffffffffffe00148 in ?? ()
(gdb) ni
0xffffffffffe0014c in ?? ()
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0xffffffffffe022bc in ?? ()
(gdb) continue
Continuing.
(gdb) di
^C
Program received signal SIGINT, Interrupt.
0xffffffffffe022bc in ?? ()
(gdb) disassemble
No function contains program counter for selected frame.
(gdb)
```

Could you tell me the reason of the difference?

aswaterman, 2023-03-17:
I don't know.  Sounds like a question for the QEMU folks.

scottj97, 2023-03-17:
You can run Spike with `-l --log-commits` to see what Spike's doing. Maybe that will help you identify where Qemu diverges.

If you have any reason to believe that Spike is doing something incorrectly, feel free to reopen.

closed 2023-03-17 by scottj97

liweiwei90, 2023-03-18:
You need use "-M spike" for riscv-tests, since the tests stop by writing to htif which is not provided by virt machine.