riscv-software-src/riscv-isa-sim #1295: Memory address 0x10000 is invalid

adamsir32, 2023-03-25:
I used the following command:
riscv64-unknown-elf-gcc -march=rv64gcv -O3 1.c 
spike --isa=RV64GCV $( which pk) a.out
The error message is:
Access exception occurred while loading payload a.out:
Memory address 0x10000 is invalid
The code is：
/* test.c */

#include <stdio.h>
#include <stdlib.h>
#include <math.h>

#define HEIGHT 64
#define LENGTH  64

int main(int argc, char *argv[])
{
  double a [HEIGHT][LENGTH];
  double b [HEIGHT][LENGTH];

  for (int i=0; i<HEIGHT; i++)
  {
    for (int j=0; j<LENGTH;j++)
    {
       a[i][j] = i+j;
    }
  }

  for (int i=0; i<HEIGHT; i++)
  {
    for (int j=0; j<LENGTH;j++)
    {
      b[i][j] = (i+j)/LENGTH;
    }
  }

  double c [HEIGHT][LENGTH] = {0};

  for (int i=0; i<HEIGHT; i++)
  {
    for (int j=0; j<LENGTH; j++)
    {
       for (int k=0; k<HEIGHT; k++)
      {
          c[i][j] += a[i][k]*b[k][j]; 
      }
    }
  }

  printf("%.6e\n", c[0][0]);

  return 0;
}

aswaterman, 2023-03-25:
It looks like you aren’t providing Spike with a correct path to pk. $(which pk) is probably returning an empty string.

adamsir32, 2023-03-25:
> It looks like you aren’t providing Spike with a correct path to pk. $(which pk) is probably returning an empty string.

I'm not quite sure about this because when I tried to use "pk" without using the "v" extension before, it would give me an error saying it couldn't find the location. However, "$ (which pk)" can run properly, and I'm encountering the same issue here. I've also tried adding "pk" to the environment variables before, but the results were not satisfactory.

omerguzelelectronicguy, 2023-03-28:
I didn't work with pk but there is a flag in the spike that can help you maybe. 
```
-m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
```
It will change you default mem adress. So you should tell the other memory regions too (the regions which you used in you elf).

adamsir32, 2023-04-01:
> I didn't work with pk but there is a flag in the spike that can help you maybe.
>
> ```
> -m<n>                 Provide <n> MiB of target memory [default 2048]
>   -m<a:m,b:n,...>       Provide memory [...]

thanks for answering,but when i uesd 
spike -m0x10000:8192,0x20000:8192 --isa=RV64GCV $(which pk) 1
Access exception occurred while loading payload 1:
Memory address 0x701c0 is invalid
and 
spike -m0x10000:8192,0x20000:8192,0x70000:8192 --isa=RV64GCV $(which pk) 1
Access exception occurred while loading payload 1:
Memory address 0x12000 is invalid
also
spike -m0x10000:16384,0x20000:16384,0x70000:16384 --isa=RV64GCV $(which pk) 1
Access exception occurred while loading payload 1:
Memory address 0x14000 is invalid
it looks bad

closed 2023-04-01 by adamsir32

omerguzelelectronicguy, 2023-04-06:
> thanks for answering,but when i uesd spike -m0x10000:8192,0x20000:8192 --isa=RV64GCV $(which pk) 1 Access exception occurred while loading payload 1: Memory address 0x701c0 is invalid and spike [...]

You can use `readelf` with `-a` flag to see which regions you will use by working the program. Then you can give all the memories that cover your executable. As far as I know, you can give more space at once to cover all the regions(2GB as I remember, you should check)instead of giving them one by one which "looks bad". 

Another solution is to give a linker file to organize your executable address according to the spike default address. In this case, you don't need to give the address boundary in the spike command.

I hope it helps.