riscv-software-src/riscv-isa-sim #1298: spike read elf wrongly that gives me an error: exception trap_illegal_instruction

omerguzelelectronicguy, 2023-03-28:
I built the spike from this commit(a6eb6c2) at currently latest.
I am trying to port coremark boom bare-metal version to my core(rv32i). After the compilation, I tried the elf with spike.
Spike jumps exception at this point:

```
core   0: 3 0x8000375c (0x00c50733) x14 0x8000713e
core   0: 0x80003760 (0x02078263) beqz    a5, pc + 36
core   0: 3 0x80003760 (0x02078263)
core   0: 0x80003764 (0x00e57663) bgeu    a0, a4, pc + 12
core   0: 3 0x80003764 (0x00e57663)
core   0: 0x80003768 (0x00007138) c.flw   fa4, 96(a0)
core   0: exception trap_illegal_instruction, epc 0x80003768
core   0:           tval 0x00007138
core   0: >>>>  trap_entry
core   0: 0x800000e0 (0xef010113) addi    sp, sp, -272
core   0: 3 0x800000e0 (0xef010113) x2  0x80003650
core   0: 0x800000e4 (0x00112223) sw      ra, 4(sp)
core   0: 3 0x800000e4 (0x00112223) mem 0x80003654 0x80003770
core   0: 0x800000e8 (0x00212423) sw      sp, 8(sp)
core   0: 3 0x800000e8 (0x00212423) mem 0x80003658 0x80003650
```
There is a compressed instruction.
However, I dont have such an instruction. This is the objdump output at that point(in memset function):
```
80003758:	00050413          	addi	x8,x10,0
8000375c:	00c50733          	add	x14,x10,x12
80003760:	02078263          	beq	x15,x0,80003784 <memset+0x40>
80003764:	00e57663          	bgeu	x10,x14,80003770 <memset+0x2c>
80003768:	0ff5f593          	andi	x11,x11,255
8000376c:	fd9ff0ef          	jal	x1,80003744 <memset>
80003770:	00c12083          	lw	x1,12(x2)
80003774:	00040513          	addi	x10,x8,0
80003778:	00812403          	lw	x8,8(x2)
```
Why spike reads that instruction wrongly? Could anyone help me to solve this? Please tell me what I need to share with you to understand the problem.

aswaterman, 2023-03-28:
Most likely, a store instruction in your program overwrote that memory location.

omerguzelelectronicguy, 2023-04-06:
I couldn't create the same error after some changes it is not solved but the issue is different so I will close for now. I can open it if I create the same error again and check whether the program writes to that address or not.

closed 2023-04-06 by omerguzelelectronicguy

omerguzelelectronicguy, 2023-04-10:
I was able to regenerate the same error and you are right. The prior instructions overwrite that instruction. That's why it is an illegal instruction.
Thank you so much.