riscv-software-src/riscv-isa-sim #1304: Spike is canonicalizing the result of vfwredsum.vs instead of honouring vector masking

ThimayaPR, 2023-04-03:
spike faulting instruction : 

```
core   0: 0x0000000082dab8b4 (0xc4a49657) vfwredusum.vs v12, v10, v9, v0.t
core   0: 3 0x0000000082dab8b4 (0xc4a49657) e32 m1 l4 v12 0x2ec47c635d88f8c77ff8000000000000 c8_vstart 0x0000000000000000
```

VS arch state -  we are operating on fp16 while we are mask agnostic
```
core   0: 0x0000000082dbef54 (0x0c807cd7) vsetvli s9, zero, e16, m1, ta, ma
core   0: 3 0x0000000082dbef54 (0x0c807cd7) x25 0x0000000000000008 c8_vstart 0x0000000000000000 c3104_vl 0x0000000000000008 
```

Mask register : 

```
core   0: 0x0000000082dbed54 (0x7f88d057) vmfge.vf v0, v24, fa7
core   0: 3 0x0000000082dbed54 (0x7f88d057) c8_vstart 0x0000000000000000 e32 m2 l8 v1  0x675a4242ceb4848586ec98d90000b541 v0  0x5a4242a4b4848548b6f8b1327a3a4400 c1_fflags 0x0000000000000011
```

We can see that all the elements are masked in our case.

earlier operations on the operands :

```
core   0: 0x0000000082dbef38 (0x7e00d657) vmfge.vf v12, v0, ft1
core   0: 3 0x0000000082dbef38 (0x7e00d657) c8_vstart 0x0000000000000000 e32 m4 l14 v15 0x4242a42e8485485d20000000776db4c5 v14 0x42a42ec485485d88f0dfc3d2e0a64c97 v13 0xa42ec47c485d88f8776db4c697869243 v12 0x2ec47c635d88f8c7b004942dffff8000

core   0: 0x0000000082dbef3c (0x7f495457) vmfge.vf v8, v20, fs2
core   0: 3 0x0000000082dbef3c (0x7f495457) c8_vstart 0x0000000000000000 e32 m4 l14 v11 0xc47c63d088f8c7a1776db4c67fc00000 v10 0x7c63d0a9f8c7a152f0dfc3d27e9779ae v9  0x63d0a95fc7a152be7ff80000776db4c5 v8  0xd0a95f7ca152bef9776db4c6f92e0000 c1_fflags 0x0000000000000011
```

V12 - `2ec47c63 5d88f8c7 b004942d ffff8000`
V10 - `7c63d0a9 f8c7a152 f0dfc3d2 7e9779ae`
V9  - `63d0a95f c7a152be 7ff80000 776db4c5`

from section 14.4 : 
`vfwredosum.vs handlesi nactive elements and NaN payloads analogously to vfredosum.vs; vfwredusum.vs does so analogously  to vfredusum.vs.`

 
section 14.3: 
`If no elements are active, no additions are performed, so the scalar in vs1[0] is simply copied to the destination register, without Note canonicalizing NaN values and without setting any exception flags. This behavior preserves the handling of NaNs, exceptions, and
rounding when autovectorizing a scalar summation loop.
`

We would expect that V12 - `2ec47c63 5d88f8c7 7ff80000 776db4c5`
where we load the v12[0] with an undisturbed value from v[9]. 

Spike seems to be canonicalising the NaN in vs1[0]. 

Request for comments.

scottj97, 2023-04-03:
cc @chihminchao

aswaterman, 2023-04-03:
The remark "If no elements are active, no additions are performed" is about ordered sum reductions, not unordered sum reductions.

Regarding unordered sum reductions, the spec says "A node where one input is derived only from elements masked-off [...] may treat that input as the additive identity" and later "As a consequence of this definition [...] if no elements are active and the scalar input is NaN, implementations are permitted to canonicalize the NaN".

So, it sounds like Spike is doing something valid here.

mlawson-tt, 2023-04-04:
We actually just encountered this same situation recently and came to the conclusion that Spike's canonicalizing of the result in the case where there are no active elements for `vfredsum.vs` and `vfwredsum.vs` is valid according to the spec.  It isn't canonicalizing the result for `vfredosum.vs` or `vwredosum.vs` (as it's not allowed to do so), so Spike doesn't seem to be necessarily "wrong" here.

It seems like it'd be nice if this behavior was configurable so that Spike won't mismatch with implementations that do not canonicalize the results of these instructions though (maybe it already is?) since that option also appears to be valid according to the spec.

aswaterman, 2023-04-04:
There's so much flexibility in the implementation of the unordered sum reductions that providing configurability for the canonicalization case seems to me to address only a tiny sliver of the problem.  It doesn't seem practical to make configurable the reduction tree shape and the accumulator precision, for example.  It might be more appropriate for vendors to manually modify Spike when they need particular behavior here.

mlawson-tt, 2023-04-04:
Seems like a decent option to me.  Conveniently in this particular case I think it's just a simple bool value change in a couple of files to get the ordered canonicalization behavior, so it doesn't seem like a huge hassle to me to just make a local modification if you need the other behavior (especially since anything more complicated would probably need a local modification anyway as you mention).

closed 2023-06-15 by ThimayaPR