riscv-software-src/riscv-isa-sim #1310: Spike should fault at vstart > vlmax for vle<n>ff

ajayvreddy, 2023-04-07:
We are running with vlen=128, xlen-64.
from the spec and this issue filed on riscv-spec: https://github.com/riscv/riscv-v-spec/issues/871

for vstart>vlmax, spike should obey:
`The use of vstart values greater than the largest element index for the current SEW setting is reserved.`

Here is a snippet of spike log:  The instruction we are expecting to see a fault : vle8ff.v

```
core   0: 0xffffffd042665ac8 (0x05a07057) vsetvli zero, zero, e64, m4, ta, mu
core   0: 0 0xffffffd042665ac8 (0x05a07057) c8_vstart 0x0000000000000000 c3104_vl 0x0000000000000008 c3105_vtype 0x000000000000005a
core   0: 0xffffffd042665acc (0x05d90607) vluxei8.v v12, (s2), v29, v0.t
core   0: 0 0xffffffd042665acc (0x05d90607) c8_vstart 0x0000000000000000 e64 m4 l8
core   0: 0xffffffd042665ad0 (0x0084d073) csrwi   vstart, 9
core   0: 0 0xffffffd042665ad0 (0x0084d073) c8_vstart 0x0000000000000009
core   0: 0xffffffd042665ad4 (0x100832af) lr.d    t0, (a6)
core   0: 0 0xffffffd042665ad4 (0x100832af) x5  0x000000000000e200 mem 0x000000148479f970
core   0: 0xffffffd042665ad8 (0x57f0f593) andi    a1, ra, 1407
core   0: 0 0xffffffd042665ad8 (0x57f0f593) x11 0x000000000000015c
core   0: 0xffffffd042665adc (0xe0098fd3) fmv.x.w t6, fs3
core   0: 0 0xffffffd042665adc (0xe0098fd3) x31 0x000000005f800000
core   0: 0xffffffd042665ae0 (0xb06a1da3) sh      t1, -1253(s4)
core   0: 0 0xffffffd042665ae0 (0xb06a1da3) mem 0x0000001904dab96b 0x0009
core   0: 0xffffffd042665ae4 (0xd2338ed3) fcvt.d.lu ft9, t2
core   0: 0 0xffffffd042665ae4 (0xd2338ed3) c1_fflags 0x0000000000000019 f29 0x43f0000000000000
core   0: 0xffffffd042665ae8 (0x01090507) vle8ff.v v10, (s2), v0.t
core   0: 0 0xffffffd042665ae8 (0x01090507) c8_vstart 0x0000000000000000 e64 m4 l8

 
```

aswaterman, 2023-04-07:
Trapping on reserved instructions is recommended but not required.  So I'd rather leave this alone.

ajayvreddy, 2023-04-07:
I see that it is taking trap for few other instruction (see example of vslidedown below).
Can Spike be consistent? or have a switch which the user can decide on whether to trap or not ?

```
core   0: 0x0000000082dc5d94 (0x00967657) vsetvli a2, a2, e16, m2, tu, mu
core   0: 3 0x0000000082dc5d94 (0x00967657) x12 0x0000000000000010 c8_vstart 0x0000000000000000 c3104_vl 0x0000000000000010 c3105_vtype 0x0000000000000009
core   0: 0x0000000082dc5d98 (0xc20025f3) csrr    a1, vl
core   0: 3 0x0000000082dc5d98 (0xc20025f3) x11 0x0000000000000010
core   0: 0x0000000082dc5d9c (0x00158593) addi    a1, a1, 1
core   0: 3 0x0000000082dc5d9c (0x00158593) x11 0x0000000000000011
core   0: 0x0000000082dc5da0 (0x00010637) lui     a2, 0x10
core   0: 3 0x0000000082dc5da0 (0x00010637) x12 0x0000000000010000
core   0: 0x0000000082dc5da4 (0x5536061b) addiw   a2, a2, 1363
core   0: 3 0x0000000082dc5da4 (0x5536061b) x12 0x0000000000010553
core   0: 0x0000000082dc5da8 (0x00f61613) slli    a2, a2, 15
core   0: 3 0x0000000082dc5da8 (0x00f61613) x12 0x0000000082a98000
core   0: 0x0000000082dc5dac (0x15060613) addi    a2, a2, 336
core   0: 3 0x0000000082dc5dac (0x15060613) x12 0x0000000082a98150
core   0: 0x0000000082dc5db0 (0x02065f27) vse16.v v30, (a2)
core   0: 3 0x0000000082dc5db0 (0x02065f27) c8_vstart 0x0000000000000000 e16 m2 l16 mem 0x0000000082a98150 0xaa04 mem 0x0000000082a98152 0x0800 mem 0x0000000082a98154 0x00a4 mem 0x0000000082a98156 0x0000 mem 0x0000000082a98158 0x005e mem 0x0000000082a981core   0: 0x0000000082dc5db4 (0x0000061b) sext.w  a2, zero                                                                                                                                                                                                    core   0: 3 0x0000000082dc5db4 (0x0000061b) x12 0x0000000000000000
core   0: 0x0000000082dc5db8 (0x00859073) csrw    vstart, a1
core   0: 3 0x0000000082dc5db8 (0x00859073) c8_vstart 0x0000000000000011
core   0: 0x0000000082dc5dbc (0x3ecccf57) vslidedown.vx v30, v12, s9
core   0: exception trap_illegal_instruction, epc 0x0000000082dc5dbc
core   0:           tval 0x000000003ecccf57
```

aswaterman, 2023-04-07:
That's actually a different case: as permitted by the ISA, Spike traps when vstart is nonzero on arithmetic instructions (regardless of whether vstart < vlmax).

closed 2023-04-09 by ajayvreddy