riscv-software-src/riscv-isa-sim #1323: LR/SC: Spike dropping reservation

ajayvreddy, 2023-04-15:
[image]

I see SC is not getting succeded (as above) although it satisfies the constrained eventual success .

Is is possible for Spike to print out the message when it drops a reservation with appropriate reason.

aswaterman, 2023-04-17:
In this case, it's probably because Spike drops reservations periodically when it switches to modeling another hart.  (This happens even if there's only one hart.)

Printing out that information sounds like something you should do on your local branch.  You can instrument the `mmu_t::yield_load_reservation` function, or instrument the call sites. 
 It's only called from a small number of places.

iamkarthikbk, 2024-05-27:
increasing [this](https://github.com/riscv-software-src/riscv-isa-sim/blob/a53a71fcc3c985cf95973e86e40814c30c551a68/riscv/sim.h#L67) number locally and recompiling spike locally might help.