riscv-software-src/riscv-isa-sim #1324: VFMADD: incorrect decoding.

ajayvreddy, 2023-04-16:
I am running spike with the following options:  `--isa=rv64imafdcv_zfh_zba_zbb_zbc_zbs --misaligned --priv=msu --varch=vlen:128,elen:64
`

I am running a test with vectors, from disassembly :
` 80001922:a22a1457          vfmadd.vvv8,v20,v2`

but from spike log (vs1, vs2 are interchanged):
```
core   0: 0x0000000080001922 (0xa22a1457) vfmadd.vv v8, v2, v20
core   0: exception trap_illegal_instruction, epc 0x0000000080001922
core   0:           tval 0x00000000a22a1457

```

scottj97, 2023-04-20:
Without knowing the vector spec in detail, this could be a binutils (assembler/disassembler) problem, and/or a Spike disassembler problem, and/or a Spike semantic problem. It would be useful if you could show us the expected field encoding from the Vector spec, as well as the original assembly source (if there is such a thing), so we can see where the discrepancy lies.

It would also be useful to run some instructions that succeed instead of trap, using simple arithmetic, so we can tell if Spike is swapping vs1/vs2 in its execution, or only in the displayed disassembly.

WojciechMula, 2023-07-13:
The Spike disassembler follows the RVV spec. A simple FP instruction is defined as:

```
# Floating-point add
vfadd.vv vd, vs2, vs1, vm # Vector-vector
```

For `fmadd` (and friends) we vs1 and vs2 are swap.

```
# FP multiply-add, overwrites multiplicand
vfmadd.vv vd, vs1, vs2, vm # vd[i] = +(vs1[i] * vd[i]) + vs2[i]
```

ajayvreddy, 2023-08-07:
Sorry for the late reply,
here is the encoding:

1. vfmadd.vv vd, vs1, vs2, vm
2. vfmadd.vv      31..26=0x28 vm vs2 vs1 14..12=0x1 vd 6..0=0x57 (https://github.com/riscv/riscv-opcodes/blob/master/rv_v#L198C1-L198C62)
3. opcode: (0xa22a1457) -> vd=8, vs1=20, vs2=2,

`core   0: 0x0000000080001922 (0xa22a1457) vfmadd.vv v8, v2, v20`

`core   0: exception trap_illegal_instruction, epc 0x0000000080001922`

`core   0:           tval 0x00000000a22a1457`

Spike interchanges vs1 and vs2.

I have attached the elf file (zipped by tar -zvcf)

[risc.elf.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/12278969/risc.elf.zip)

Cmdline: `./spike   --isa=rv64imafdchv_zba_zbb_zbc_zbkb_zbkx_zbs_zknd_zkne_zknh_zksed_zksh_svinval_svnapot_svpbmt_zicbom_zicboz_zicond --priv=msu --misaligned --varch=vlen:256,elen:64 --log=xyz.log  <elf>`

aswaterman, 2023-08-07:
It looks like maybe Spike is disassembling the instruction incorrectly. I don’t think there’s a semantic problem.

ajayvreddy, 2023-08-08:
yes,  thats right.

closed 2023-08-09 by ajayvreddy