riscv-software-src/riscv-isa-sim #1328: msstatus.gva bit is still enabled on a page fault exception in M-mode when handling a VS page fault

jgauchola, 2023-04-19:
Hi,

I am having the following issue with Spike related to GVA bit. Of course, I am testing the hypervisor extension.
The following environment describes the GVA behavior with the latest released version of Spike:

```
  li t0, ((MSTATUS_MPP & ~(MSTATUS_MPP<<1)) * PRV_S) | MSTATUS_MPRV | MSTATUS_MPV | MSTATUS_MXR
  csrw mstatus, t0      # these bits enables two-stage data address translation in VS-mode with MXR enabled.
  
  sd t1, 0x0(t1)        # Makes a store using a bad GVA, raising a store page fault

  # Entering in the trap routine in M-mode
  la t0,   mhandler     # changing to a new trap routine
  csrw    mtvec, t0
  csrr    t0, mcause    # Value: 0x000000000000000f
  csrr    t1, mstatus   # Value: 0x0000004a000a1800 (GVA enabled, that's OK)
  
  li t2, 0x2000
  ld  t2, 0x0(t2)        # It gives a load access fault, as the address used is incorrect (reserved for debug mode)

  # Entering in the new trap routine mhandler in M-mode
  csrr    t0, mcause    # Value: 0x0000000000000005
  csrr    t1, mstatus   # Value: 0x0000004a000a1800 (The GVA is enabled, why?)
```

So, my question is, **why in the last trap handler _(mhandler)_ is the GVA enabled?**  It is not clear to me. 

According to Spec, GVA is set to 1 for any trap that writes a guest virtual address to _mtval_ register. But in the last load, it uses a machine address because the _mstatus_ register doesn't have enabled the virtual translation caused by handling a trap, right?

[image]

[image]
_NOTE: The current privilege mode is M-mode as only the data address translation is enabled by setting the corresponding bits in mstatus_

[image]

scottj97, 2023-04-19:
I can't explain that.

Can you share the ELF that you ran? Or at least, show us the Spike commit log around the failure (`-l --log-commits`)

elYuma, 2023-04-20:
Hi scottj97,

Here's the ELF file that I run.
[dsv39-p-vs-gva.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/11284027/dsv39-p-vs-gva.zip)

Thanks!

elYuma, 2023-04-20:
Hi, 

I think where is the misunderstanding from my side. 
When vsatp and hgatp register are enabled (mode != bare), is the two-address translation and protection enabled by default? even the core is in machine mode?

scottj97, 2023-04-20:
> When vsatp and hgatp register are enabled (mode != bare), is the two-address translation and protection enabled by default? even the core is in machine mode?

No

scottj97, 2023-04-20:
I need more info. I ran the ELF you provided like so:
```
spike --isa=rv64gch -m0x800000000000:0x1000000 -l --log-commits dsv39-vs-gva.elf
```

The first time we hit `mhandler`, `mcause` is 7 (store access fault); after that the test runs away after just a few more instructions with endless `ecall`s going back to `mhandler`.

I never see any entry to `mhandler` with `mcause` of 5 as you observed in the original post above.

Can you share your command line, and any other info I might need to recreate this?

elYuma, 2023-04-24:
Hi scottj97,

It seems to be our fault. When running Spike alone it gives correct values. However, when Spike is simulated with our core, it gives incongruent values.

So I think Spike is doing fine. Probably there is an error in the cosimulation.

Anyway, thanks for your help!

aswaterman, 2023-04-24:
Thank you for reporting back!

closed 2023-04-24 by aswaterman