riscv-software-src/riscv-isa-sim #1331: why is `-g` always enabled for spike

poemonsense, 2023-04-19:
I notice from the configure script that, if the compiler accepts `-g`, the Makefile always comes with a `-g` argument to the compiler. Correct me if I'm missing something.

Why does spike do so? I think without `-g` spike could run faster.

sifiverobert, 2023-04-19:
-g only adds DWARF debug to ELF file. It should not affect generated code at all.

poemonsense, 2023-04-20:
It produces a quite larger elf file. I think it's better to have a configure option to allow the user to determine whether they want `-g`?

aswaterman, 2023-04-20:
That capability already exists: `../configure CFLAGS=-g0`

aswaterman, 2023-04-20:
This version suffices, so I'll close the issue as complete: `../configure CXXFLAGS=-g0`

(It only affects the C++ objects, but that's the vast majority.  Additionally adding `CFLAGS=-g0` to suppress debug symbols in the C objects has only a tiny effect.)

closed 2023-04-20 by aswaterman

poemonsense, 2023-04-20:
Passing `-g0` from compiling flags works. Thanks.